matth-x/MicroOcpp #1: Stand alone ESP8266 example

goRaspy, 2020-07-15:
Thank you so much for your work here.
It is very interesting that you prove the concept of a working implementation of OCCP on ESP8266.

Unfortunately I am not able to make my own tests due of missing external charging point. 
I think, it should be very helpful to dispose of a working example of a stand alone ESP8266 without external hardware.

The example may "just" do the following sequence : 
On boot
- Open web sockets
- Boot notification 
- Send a status notification on a "virtual" connector 1 

On loop : 
- Ping pong for the web socket
- Heartbeat

This example probably needs a specific EVSE.h.

Any help will be highly appreciated.

matth-x, 2020-07-15:
Hey, thank you for your input!

Actually, the two files in the example folder are a full test routine which runs without external hardware. Just execute the code on your ESP8266 and see in the console what happens! In the test routine, the ESP8266 establishes a connections to a websocket-echo server (`wss://echo.websocket.org`) and performs a full (virtual) EV plug, power scheduling und unplugging cycle.

Before compiling, you have to adapt lines 41 and 42 in example_client.cpp according to your WiFi:
`#define STASSID "YOUR_WLAN_SSID"`
`#define STAPSK  "YOUR_WLAN_PSWD"`

goRaspy, 2020-07-15:
Thank you so much for your fast feedback.

Unfortunately I was not able to compile. Do you use Arduino IDE to build ? 

Will try again soon.

matth-x, 2020-07-17:
Mostly PlatformIO, for testing I also use Arduino IDE. What's the error message?

goRaspy, 2020-07-25:
Hello, sorry for long time answer. I'll look further later. But I would like to ask you if you should be interested to work on an ocpp implementation on a son off device (esp8266 based) ? I can imagine remunerate you.

Many thanks

matth-x, 2020-07-25:
I would be excited to hear more about your project idea! Sure, provide me with contact details so you can tell me more

goRaspy, 2020-07-25:
Awesome. Please send me a short email to romain "at" farelle.com

closed 2020-09-02 by matth-x

vikramec08, 2022-05-05:
hi , 
i have tried your code and getting error 
Regards