matth-x/MicroOcpp #4: ESP32 port

Fyod, 2021-05-02:
Hi Matth-x,
I would like to test this library on ESP32, byt an getting errors.
Any tips on what would be needed to port it? 

Thanks!

RealOAAS, 2021-05-02:
Hi Matth-X, 

I'm wondering in regards to the same thing, are there any plans to port the library to ESP32 soon?

Thanks!

matth-x, 2021-05-02:
Hi,

To be honest, I already use it with the ESP32 but just haven't updated the example section yet. Have you tried including
```cpp
/* ESP32 WiFi libraries */
#include <WiFi.h>
#include <WiFiMulti.h>
#include <WiFiClientSecure.h>
```
instead of
```cpp
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
```
?

What does the compiler output say?

Hope I could help you.

TechOverflow, 2021-05-02:
The first issue was that when I saved the zip int Arduino libraries, it would not show up in the File -> Examples.
So I just created a new sketch and copied over the example.

The includes are like this

```
#include <Variants.h>
#include <Arduino.h>
#include <ArduinoJson.h>
#include <WiFi.h>
#include <WiFiMulti.h>
#include <ESP8266-OCPP.h>
#include "OneConnector_HW_integration.h"
```

Cheers!

Fyod, 2021-05-02:
The first issue was that when I saved the zip int Arduino libraries, it would not show up in the File -> Examples.
So I just created a new sketch and copied over the example.

The includes are like this

```
#include <Variants.h>
#include <Arduino.h>
#include <ArduinoJson.h>
#include <WiFi.h>
#include <WiFiMulti.h>
#include <ESP8266-OCPP.h>
#include "OneConnector_HW_integration.h"
```

The errors are

```
/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:162:41: error: declaration of 'T ArduinoOcpp::Configuration<T>::operator=(T) noexcept' has a different exception specifier
 T Configuration<T>::operator=(T newVal) noexcept {

In file included from /Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:5:0:
Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.h:61:7: error: from previous declaration 'T ArduinoOcpp::Configuration<T>::operator=(T)'
     T operator=(T newVal);

/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp: In function 'bool ArduinoOcpp::configuration_init()':
/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:752:5: error: 'SPIFFSConfig' was not declared in this scope
     SPIFFSConfig cfg;

/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:753:5: error: 'cfg' was not declared in this scope
     cfg.setAutoFormat(true);

...
```

Cheers!

matth-x, 2021-05-03:
Okay, those messages look kind of familiar to me ... can you download the branch "develop" instead of "master" and try it again? Unfortunately, the persistent storage does not work for the ESP32 at the moment and I will migrate to LittleFS in the long run. You have to manually turn it off by setting the build flag `AO_DEACTIVATE_FLASH` (the Readme in https://github.com/matth-x/ArduinoOcpp/tree/master/examples/ESP32 explains it).

Furthermore, please make sure that your IDE uses version `6.17.2` of ArduinoJson. You might have to downgrade it.

I have added an [example file for the ESP32](https://github.com/matth-x/ArduinoOcpp/blob/master/examples/ESP32/main.cpp). I hope it works for you. Please let me know if everything worked and btw., thank you for your feedback!

Fyod, 2021-05-04:
So I downloaded the development version and tried the ESP32 example file, still getting some errors, but the `T operator` error is gone.

```
/Users/xxx/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp: In function 'bool ArduinoOcpp::configuration_init()':
/Users/xxx/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:753:5: error: 'SPIFFSConfig' was not declared in this scope
     SPIFFSConfig cfg;
     ^
/Users/xxx/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:754:5: error: 'cfg' was not declared in this scope
     cfg.setAutoFormat(true);
     ^
/Users/xxx/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:755:5: error: 'SPIFFS' was not declared in this scope
     SPIFFS.setConfig(cfg);
     ^
/Users/xxx/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp: In function 'bool ArduinoOcpp::configuration_save()':
/Users/xxx/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:895:5: error: 'SPIFFS' was not declared in this scope
     SPIFFS.remove(CONFIGURATION_FN);
     ^

exit status 1
Error compiling for board ESP32 Dev Module.

```
Not of concern:

```
WARNING: library ArduinoOcpp claims to run on espressif8266 architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).

Multiple libraries were found for "WiFi.h"
 Used: /Users/xxx/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
```

matth-x, 2021-05-04:
The library is not compatible with the SPIFFS implementation on the ESP32 yet. The workaround is to delete the code which uses SPIFFS commands. Basically, if you delete everything inside the precompiler statements

```cpp
#ifndef AO_DEACTIVATE_FLASH
    ... //code which uses unsupported SPIFFS library. Delete this
#endif //ndef AO_DEACTIVATE_FLASH
```

you should be fine. They are in `src/ArduinoOcpp/Core/Configuration.cpp` and `src/ArduinoOcpp/Tasks/SmartCharging/SmartChargingService.cpp`.

Fyod, 2021-05-05:
That worked, thanks!

matth-x, 2021-05-05:
That's nice to hear! If you encounter more problems, please don't hesitate to report them.

closed 2021-05-05 by matth-x

matth-x, 2021-05-10:
@lincomatic Thank you, your input is much appreciated. Now the `develop` branch runs directly on the ESP32 without editing something before.