matth-x/MicroOcpp #10: Littlefs not finding default configuration

Joelmaca9, 2021-07-05:
Hi!!

I´m trying to build your proyect in an ESP32 withthe esp-idf. I´m able to build and flash it, but i have a problem when loading the default configuration file. I changed my partitions file to create a spiffs directory and I created an arduino-ocpp.cnf file, but of course it is empty and im not able to find in your repository that file. 

thanks in advance for your help and your work with this amazing repo!

fanzypantz, 2021-07-06:
I'm having some Littlefs issues too, but probably a dumber mistake on my part. It simply can't find it: 

`.pio\libdeps\esp32dev\ArduinoOcpp\src\ArduinoOcpp\Core\Configuration.cpp:16:22: fatal error: LITTLEFS.h: No such file or directory`

More similar errors are occuring. Trying to build your ESP32 example.

matth-x, 2021-07-06:
Hi,

@Joelmaca9 This library does not need a preexisting configs file. When the configs file is missing, it loads the defaults from the program memory and it will create the configs file in the flash file system as soon as the first entry changes. You can trigger that by sending an OCPP `ChangeConfiguration` request to your ESP32 or by starting a transaction. Alternatively, including `<ArduinoOcpp/Core/Configuration.h>` and calling `ArduinoOcpp::configuration_save()` anywhere after initializing the library will do the same. Can you try it and check the FS contents?

@fanzypantz Can you post the library dependency graph of the pio build routine? But my first guess is that copying the example `platformio.ini` into the root project folder and selecting the environment `esp32-development-board` could be worth a try.

Joelmaca9, 2021-10-05:
Sorry for not responding but I have been very occupied. I had a lot of troubles with littlefs so I ended up using the spiffs system and now works great! Thank you very much.

Also I wanted to know if you have any memory system for the authorization caché. I have been looking around the code but I have not seen anything and I need to implement it both with the Cear Cache function.

YehorFedorov, 2022-04-01:
Hi @Joelmaca9!
Did you manage to get this project work with esp-idf?
Could you share it to public? Or maybe some advices to get this done?

Appreciate!

closed 2023-01-07 by matth-x