matth-x/MicroOcpp #11: WSS not supported on ESP32

estebancere, 2021-08-18:
Hi!

First of all, thanks for your work! It's been very helpfull to us. 

We are developing an EVSE using ESP32 and we need to communicate with a CS using wss. This library uses a websocket library which does not support wss comunication, but we found that it is possible to use another websocket library instead of links2004/WebSockets library here:

`#ifndef AO_CUSTOM_WS
//uses links2004/WebSockets library
void OCPP_initialize(String CS_hostname, uint16_t CS_port, String CS_url, float V_eff = 230.f /*German grid*/, ArduinoOcpp::FilesystemOpt fsOpt = ArduinoOcpp::FilesystemOpt::Use_Mount_FormatOnFail, ArduinoOcpp::OcppClock system_time = ArduinoOcpp::Clocks::DEFAULT_CLOCK);
#endif

//Lets you use your own WebSocket implementation
void OCPP_initialize(ArduinoOcpp::OcppSocket *ocppSocket, float V_eff = 230.f /*German grid*/, ArduinoOcpp::FilesystemOpt fsOpt = ArduinoOcpp::FilesystemOpt::Use_Mount_FormatOnFail, ArduinoOcpp::OcppClock system_time = ArduinoOcpp::Clocks::DEFAULT_CLOCK);`

How can we use another websocket library? Could you help us?

matth-x, 2021-08-19:
Hi estebancere,

Thanks! You are right, you can provide a custom WS socket to ArduinoOcpp and therefore use any WS library which runs on your platform. A great example for that is OpenEVSE which uses the Mongoose WebSockets library. You can find the WS wrapper here: [MongooseOcppSocketClient.cpp](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/src/MongooseOcppSocketClient.cpp).

But maybe it's not necessary for you to exchange the WS library. links2004 has indeed built in most standard cryptographic functions into his library. And I plan to extend the ArduinoOcpp API with direct access to them soon. In the meantime, you can configure the WebSockets instance by yourself. Take a look at how `OCPP_initalize(...)` works now ([see here](https://github.com/matth-x/ArduinoOcpp/blob/master/src/ArduinoOcpp.cpp#L104)). You can basically copy its contents into your `setup()` function and switch from `websocket.begin(...)` to `websocket.beginSslWithCA(...)`. links2004 provides great examples on how to use WSS on his GitHub page.

Hope I could help

estebancere, 2021-08-31:
Hi Matthias!

It worked just fine! We been able to connect to the CS. Now trying to do some operations. Thanks a lot again!