matth-x/MicroOcpp #12: Energy sampler issue

fanzypantz, 2021-08-25:
This is more of a question than  an issue. I'm testing this some more and I am currently getting:

`[ConnectorMeterValuesRecorder] Called readEnergyActiveImportRegister(), but no energySampler or handling strategy set!`

Is this a setup error of the code from ArduinoOCPP or from my OCPP test  server? I'm using the HW_integration example with some changes, mostly just within the loop. 
I'm currently using the python library to test this.

The OCPP server for testing if it helps:

```python
import asyncio
import logging
from datetime import datetime

try:
    import websockets
except ModuleNotFoundError:
    print("This example relies on the 'websockets' package.")
    print("Please install it by running: ")
    print()
    print(" $ pip install websockets")
    import sys
    sys.exit(1)

from ocpp.routing import on
from ocpp.v16 import ChargePoint as cp
from ocpp.v16.enums import Action, RegistrationStatus
from ocpp.v16 import call_result

logging.basicConfig(level=logging.DEBUG)

class ChargePoint(cp):
    @on(Action.BootNotification)
    def on_boot_notification(self, charge_point_vendor: str, charge_point_model: str, **kwargs):
        return call_result.BootNotificationPayload(
            current_time=datetime.utcnow().isoformat(),
            interval=10,
            status=RegistrationStatus.accepted
        )

    @on(Action.Authorize)
    def on_authorize(self, id_tag: str):
        return call_result.AuthorizePayload(
            id_tag_info={"status": 'Accepted'},
        )

    @on(Action.StatusNotification)
    def on_status_notification(self, connector_id: str, error_code: str, status: str, timestamp: str):        
        return call_result.StatusNotificationPayload()

    @on(Action.MeterValues)
    def on_meter_value(self):
        return call_result.MeterValuesPayload()

    @on(Action.StartTransaction)
    def on_start_transaction(self, connector_id, id_tag, timestamp, meter_start):
        return call_result.StartTransactionPayload(
            id_tag_info={
                "status": 'Accepted'
            },
            transaction_id=int(1)
        )

    @on(Action.StopTransaction)
    def on_stop_transaction(self, transaction_id, timestamp, meter_stop):
        return call_result.StopTransactionPayload()

async def on_connect(websocket, path):
    """ For every new charge point that connects, create a ChargePoint
    instance and start listening for messages.
    """
    try:
        requested_protocols = websocket.request_headers[
            'Sec-WebSocket-Protocol']
    except KeyError:
        logging.error(
            "Client hasn't requested any Subprotocol. Closing Connection"
        )
        return await websocket.close()
    if websocket.subprotocol:
        logging.info("Protocols Matched: %s", websocket.subprotocol)
    else:
        # In the websockets lib if no subprotocols are supported by the
        # client and the server, it proceeds without a subprotocol,
        # so we have to manually close the connection.
        logging.warning('Protocols Mismatched | Expected Subprotocols: %s,'
                        ' but client supports  %s | Closing connection',
                        websocket.available_subprotocols,
                        requested_protocols)
        return await websocket.close()

    charge_point_id = path.strip('/')
    cp = ChargePoint(charge_point_id, websocket)

    await cp.start()

async def main():
    server = await websockets.serve(
        on_connect,
        '0.0.0.0',
        8834,
        subprotocols=['ocpp1.6']
    )

    logging.info("Server Started listening to new connections...")
    await server.wait_closed()

if __name__ == '__main__':
    try:
        # asyncio.run() is used when running this example with Python 3.7 and
        # higher.
        asyncio.run(main())
    except AttributeError:
        # For Python 3.6 a bit more code is required to run the main() task on
        # an event loop.
        loop = asyncio.get_event_loop()
        loop.run_until_complete(main())
        loop.close()

```

matth-x, 2021-08-25:
Everything is right on your side. This console message is only a reminder that you still have to give the library an access method to the energy meter of the charge point. And it is no indication of an error in your backend.

You can disable these kinds of messages by removing the build flag `-DAO_DEBUG_OUT` from the `platformio.ini`

fanzypantz, 2021-08-25:
Thanks for the quick answer. I'm trying to connect it up with a modbus meter.  Following the "OneConnector-EVSE" again, I assume what I'd do is  change the EVSE_readChargeRate() to something along the lines of this?

```c++
float EVSE_readChargeRate() { //estimation for EVSEs without power meter
    if (evIsPlugged && evRequestsEnergy) { //example
        return lastModbusReading;
    } else {
        return 0.0f;
    }
}
```

For something as simple as this, a plug with a meter on it, a fake car charger if you must. Does the "OneConnector-EVSE" need some different access or configuration? (assuming the setup is the same as example)

matth-x, 2021-09-14:
Hi fanzypanth, did you succeed? Yes that's how `EVSE_readChargeRate()` is supposed to work.

If you mean the energy and power readings, then no, it hardly needs any configuration. Just provide the raw data via `EVSE_readChargeRate()` and the library is fine. 

To not only integrate the power reading but also the energy reading, please add

```cpp
setEnergyActiveImportSampler([]() {
    return lastModbusReadingWh;
});
```

anywhere in your `setup()` function.

fanzypantz, 2021-09-14:
Right yeah, I got this library to run with the test server, also with providing metering data. Was an issue mostly with my OCPP master server.

closed 2021-09-14 by fanzypantz