matth-x/MicroOcpp #13: Build problem

gazdela12, 2021-09-04:
Hello, I try to upload occp on ESP32 but it can't even build. 
Here is screenshot, could you please help me?
[image]

204maker, 2021-09-23:
I completely agree. 
I am having errors building the code in Visual Studio Code on PlatformIO.
May I ask what version release is working at the moment?
How about a more detailed instruction on setting the environment for a successful build in Visual Studio Code on PlatformIO?

matth-x, 2021-09-25:
The error message means that the linker doesn't find the compiled versions of the `loop()` and `setup()` function. Somehow the compiler doesn't compile them. Have you copied the example files into the source folder of your project?

I use PIO core version `5.1.1`.

gazdela12, 2021-10-03:
I managed to fix it but now I got this:
[image]

HackLink123, 2021-12-16:
Hi gazdela12  I met the same problem as you.How did you solve it?

matth-x, 2022-02-16:
@gazdela12 The issue here was that `echo.websocket.org` is down unfortunately. I've updated the URL in the example sketch.

closed 2022-02-16 by matth-x

Navjot0005, 2023-09-01:
Hi @matth-x 
I am facing the exact same problem.
How I can troubleshoot this??
Do I need to make a new project ??
Because I did not make.
What I am currently doing unzip the downloaded folder and without any change opening in PlatformIO.
I did not make new project.
Can you guide me how shall I start this because I am not very good familiar with PlatformIO. ?
Shall I made new project in PlatformIO?
if I made new then how I add MicroOcPP in the new project?
Any help will be appreciated.
Waiting for reply..