matth-x/MicroOcpp #14: Arduino IDE setup process

Pulse-akhil, 2021-09-06:
Hi Matth, I am trying to setup your github repo within my Arduino repo. Is there a wiki or a step by step instruction on how one can setup the OneConnector_EVSE.ino file on arduino IDE? Very much appreciate any help you can offer here.

matth-x, 2021-09-14:
Hi Akhil, you can copy the whole repo into the Arduino IDE library folder. To get started, please create a new project and copy all files from the OneConnector_EVSE folder into the project folder.

Pulse-akhil, 2021-09-18:
Thank you Matth-x. Let me give this a go. I have successfully setup Steve. Now time to get a chargepoint simulator running using your code

204maker, 2021-09-23:
> Hi Akhil, you can copy the whole repo into the Arduino IDE library folder. To get started, please create a new project and copy all files from the OneConnector_EVSE folder into the project folder.

Are there any special tweaks to be made to be able to compile/run successfully in Arduino for ESP32?
I am having errors with missing files.

matth-x, 2021-09-25:
At this moment, only the example in the folder `ESP32` can compile for the ESP32. But it showcases all mechanics of the OCPP library, so it is a good starting point as well.

You can patch the other examples by adapting the Wi-Fi connection method in the main file and changing the pin mapping. And it needs the dependencies from the ESP32 environment in the `platformio.ini`.

Pulse-akhil, 2021-10-14:
Resolved the above issue by doing two things
a)  editing file "ConfigurationContainerFlash.cpp' inside the ArduinoOCpp library. The edit was to simply replace any reference for LITTLEFS to LittleFS
b) installed the esp32 board libraries as well. 

After compiling post the steps above, I get a new error as follows.
 
WARNING: library ArduinoOcpp claims to run on espressif8266, espressif32 architecture(s) and may be incompatible with your current board which runs on esp8266 architecture(s).
/Users/akhil/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/ChangeConfiguration.cpp: In member function 'virtual void ArduinoOcpp::Ocpp16::ChangeConfiguration::processReq(ArduinoJson::JsonObject)':
/Users/akhil/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/ChangeConfiguration.cpp:43:32: error: invalid conversion from 'char' to 'const char*' [-fpermissive]
   43 |     const char *value_string = '\0';
      |                                ^~~~
      |                                |
      |                                char
exit status 1
Error compiling for board Generic ESP8266 Module.

EZBOT1, 2022-08-08:
> Found the issue. For this code to compile successfully, you need to change the board version from esp8266 3.0 to 2.7.4. One can do this by going to Tools -> Boards -> Boardmanager --> type in ESP8266 --> select 2.7.4 and install.

hey I changes the board version but getting new error . can you help?
`

WARNING: library ArduinoOcpp claims to run on espressif8266, espressif32 architecture(s) and may be incompatible with your current board which runs on esp8266 architecture(s).
In file included from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:37:0,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/OcppOperationCallbacks.h:8,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/OcppOperation.h:14,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/OcppOperation.cpp:5:
/Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp: In instantiation of 'static void ArduinoJson6194_F1::Converter<T, Enable>::toJson(const T&, ArduinoJson6194_F1::VariantRef) [with T = std::basic_string<char>; Enable = void]':
/Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:173:38:   required from 'bool ArduinoJson6194_F1::VariantRef::set(const T&) const [with T = std::basic_string<char>]'
/Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:201:27:   required from 'bool ArduinoJson6194_F1::JsonDocument::add(const TValue&) [with TValue = std::basic_string<char>]'
/Users/aashutoshjha/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/OcppOperation.cpp:118:36:   required from here
/Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:17:27: error: no matching function for call to 'convertToJson(const std::basic_string<char>&, ArduinoJson6194_F1::VariantRef&)'
     convertToJson(src, dst); // Error here? See https://arduinojson.org/v6/unsupported-set/
                           ^
/Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:17:27: note: candidates are:
In file included from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/BasicJsonDocument.hpp:7:0,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:28,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/OcppOperationCallbacks.h:8,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/OcppOperation.h:14,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/OcppOperation.cpp:5:
/Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:291:13: note: void ArduinoJson6194_F1::convertToJson(const ArduinoJson6194_F1::JsonDocument&, ArduinoJson6194_F1::VariantRef)
 inline void convertToJson(const JsonDocument& src, VariantRef dst) {
             ^
/Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:291:13: note:   no known conversion for argument 1 from 'const std::basic_string<char>' to 'const ArduinoJson6194_F1::JsonDocument&'
In file included from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:37:0,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/OcppOperationCallbacks.h:8,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/OcppOperation.h:14,
                 from /Users/aashutoshjha/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/OcppOperation.cpp:5:
/Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:247:13: note: void ArduinoJson6194_F1::convertToJson(const Printable&, ArduinoJson6194_F1::VariantRef)
 inline void convertToJson(const ::Printable& src, VariantRef dst) {
             ^
/Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:247:13: note:   no known conversion for argument 1 from 'const std::basic_string<char>' to 'const Printable&'
/Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:155:59: note: template<class T> typename ArduinoJson6194_F1::enable_if<ArduinoJson6194_F1::IsString<TString, void>::value, bool>::type ArduinoJson6194_F1::convertToJson(const T&, ArduinoJson6194_F1::VariantRef)
 inline typename enable_if<IsString<T>::value, bool>::type convertToJson(
                                                           ^
[... cut ...] n&) [with TDestination = std::basic_string<char>; size_t = unsigned int]'
/Users/aashutoshjha/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/OcppOperation.cpp:130:35:   required from here
/Users/aashutoshjha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Serialization/Writer.hpp:18:26: error: 'class std::basic_string<char>' has no member named 'write'
     return _dest->write(c);
                          ^
exit status 1
Error compiling for board Generic ESP8266 Module.

`