matth-x/MicroOcpp #16: Compile Error

ChaitanyaYeole02, 2021-09-18:
In file included from ~\Arduino\libraries\ArduinoOcpp\src\ArduinoOcpp\Core\Configuration.cpp:5:
~\Arduino\libraries\ArduinoOcpp\src/ArduinoOcpp/Core/Configuration.h:24:26: error: 'vector' is not a member of 'std'

td::shared_ptr<std::vector<std::shared_ptr<AbstractConfiguration>>> getAllConfigurations();
                          ^~~~~~
~\Arduino\libraries\ArduinoOcpp\src/ArduinoOcpp/Core/Configuration.h:24:26: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
~\Arduino\libraries\ArduinoOcpp\src/ArduinoOcpp/Core/Configuration.h:10:1:
+#include <vector>

~\Arduino\libraries\ArduinoOcpp\src/ArduinoOcpp/Core/Configuration.h:24:26:
     std::shared_ptr<std::vector<std::shared_ptr<AbstractConfiguration>>> getAllConfigurations();
                          ^~~~~~
~\libraries\ArduinoOcpp\src/ArduinoOcpp/Core/Configuration.h:24:26: error: 'vector' is not a member of 'std'
~\Arduino\libraries\ArduinoOcpp\src/ArduinoOcpp/Core/Configuration.h:24:26: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
~\Arduino\libraries\ArduinoOcpp\src/ArduinoOcpp/Core/Configuration.h:24:70: error: template argument 1 is invalid
     std::shared_ptr<std::vector<std::shared_ptr<AbstractConfiguration>>> getAllConfigurations();
                                                                      ^~
~\\Arduino\libraries\ArduinoOcpp\src/ArduinoOcpp/Core/Configuration.h:24:72: error: expected unqualified-id before '>' token
     std::shared_ptr<std::vector<std::shared_ptr<AbstractConfiguration>>> getAllConfigurations();

this is some vector error, i am not yet familiar with vectors and how to solve this error
                                                                        ^

ChaitanyaYeole02, 2021-09-18:
I removed #include <vector> from Configuration.cpp and included it in Configuration.h, same I did for SmartChargingModel.h and GetConfiguration.h

ChaitanyaYeole02, 2021-09-18:
~\Arduino\libraries\ArduinoOcpp\src\ArduinoOcpp\MessagesV16\ChangeConfiguration.cpp: In member function 'virtual void ArduinoOcpp::Ocpp16::ChangeConfiguration::processReq(ArduinoJson::JsonObject)':
~\Arduino\libraries\ArduinoOcpp\src\ArduinoOcpp\MessagesV16\ChangeConfiguration.cpp:43:32: error: invalid conversion from 'char' to 'const char*' [-fpermissive]
     const char *value_string = '\0';

matth-x, 2021-09-19:
Hi Chaitanya,

Probably, some dependencies aren't working yet. Have you copied the environment section `[env:esp32-development-board]` from the `platformio.ini` file into your own file?

Anytime when the compiler complains about the missing `std::vector`, please add `#include <vector>` to the file. That will help track down the error. The last compiler error looks like it expects a c-style string as initialization value, so try it by changing `const char *value_string = '\0';` to `... = "";`

ChaitanyaYeole02, 2021-09-21:
Thank you, the issue has been resolved

closed 2021-09-21 by ChaitanyaYeole02

shardul10, 2022-07-22:
I am getting these 'vector' errors in the SmartChargingService and GetConfiguration. These started appearing suddenly. Any specific reason for that? Did I accidentally upgrade/downgrade ArduinoOcpp?