matth-x/MicroOcpp #22: CALLRESULT error

ChaitanyaYeole02, 2021-10-05:
This error happens randomly, if I keep my esp on it will show this error on heartbeat, then sometimes it also shows the same error when I reboot my esp. 
@matth-x Have you faced this problem, every suggestion on this issue would be helpful
[OcppEngine] Received CALLRESULT doesn't match any pending operation!

matth-x, 2021-10-05:
Hi Chaitanya, the error says that the OCPP server has sent a response message which cannot be assigned to a pending operation.

So if your Central System doesn't alter the message IDs (which I strongly assume) the most likely reason is that the client has initiated a Heartbeat operation and the Central System has replied with two identical responses. Then the first response finishes the pending operation and the second one arrives in an empty queue causing that debug message. That's a side effect of package losses with TCP connections and how OCPP is designed to deal with it.

Another possible reason is that the response arrives at the moment when the client cancels the operation because of the timeout or the client has rebooted in the meantime. In all cases it is correct to just discard the incoming response message and do nothing more.

If you enable the `AO_TRAFFIC_OUT` you can reconstruct the message flow. I would certainly be interested if you find a further case causing that debug message!

ChaitanyaYeole02, 2021-10-06:
My OCPP server is a basic Python OCPP server, which is available at https://pypi.org/project/ocpp/

I just saw the logs after enabling AO_TRAFFIC_OUT

[OcppOperation] Sent requirement (success): [2,"531531532","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2021-10-06T06:58:35.000Z"}]
[OcppOperation] Sent requirement (success): [2,"531531532","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2021-10-06T06:58:35.000Z"}]
[OcppClientSocket] get text: [3,"531531532",{}]
[OcppClientSocket] get text: [3,"531531532",{}]

ChaitanyaYeole02, 2021-10-06:
I think the ID's from the server are the same, The problem is from Charge Point side where is it not able to respond to the CALLRESULT function from the server

ChaitanyaYeole02, 2021-10-06:
Thank you so much, I will try this and let you know