matth-x/MicroOcpp #28: Need guidance on how the code would look like for remote start transaction

Pulse-akhil, 2021-10-15:
I am trying to get our OCPP server to start the charger using RemoteStartTransaction request. Question I have here is with OCPP_loop() does the program automatically detect such a request and react to it OR would we need to enable a function to start the energy flow when onRemoteStartTransactionRecevieRequest is invoked? 

An example function on how this would look like would be very helpful.

matth-x, 2021-10-16:
Hi Akhil,

Yes, the OCPP library listens to `RemoteStartTransaction` requests and your client has to define a reaction to it. See [this example](https://github.com/matth-x/ArduinoOcpp/blob/8331cc6fb722e8bbb5b43661c93b3590f2747df6/examples/SECC/main.ino#L257).

closed 2021-11-04 by matth-x

akrokon1992, 2022-10-29:
getting error 
src/main.cpp:258:30: error: too few arguments to function 'void startTransaction(const char*, ArduinoOcpp::OnReceiveConfListener, ArduinoOcpp::OnAbortListener, ArduinoOcpp::OnTimeoutListener, ArduinoOcpp::OnReceiveErrorListener, std::unique_ptr<ArduinoOcpp::Timeout>)'