matth-x/MicroOcpp #33: How to setup the project?

adithyagurumurthy, 2021-11-14:
I am using NodeMcuv2
I have downloaded the project, and selected the moved the examples file "ESP/mail.ino" into the source file, but still I am getting errors 
[image]

How should I setup the project, I am unbale to do it.

adithyagurumurthy, 2021-11-14:
I was somehow able to resolve the error , but I see a new wierd characters in console
[image]

Is this intended ? how to correct this ?

matth-x, 2021-11-14:
Hi Adithya,

Yes, the first errors likely occured because of a version jump in the develop branch. Thank you for reporting this, should be fixed. The second screenshot just shows that the console is somehow misconfigured. Maybe your `platformio.ini` is missing the line `monitor_speed = 115200`?

adithyagurumurthy, 2021-11-15:
Thanks will look into that and try to compile!!

adithyagurumurthy, 2021-11-17:
I try to compile and the websocket are not working ,
[image]

I dont see any connection established to websocekts.
How to establish the websocket ?

HackLink123, 2021-12-16:
Hi adithyagurumurthy  I met the same problem as you.Can you tell me the solution? Thank you so much.

HackLink123, 2021-12-21:
The problem is that the  (Espressif 8266) version  too high
I solved the above problem and used the following methods:
[image]
Change you Espressif 8266 version in **2.6.3** .

matth-x, 2021-12-21:
Hi, thank you! I will adapt the `platformio.ini` according to your report.

matth-x, 2022-02-02:
@adithyagurumurthy Please refer to https://github.com/matth-x/ArduinoOcpp/issues/36#issuecomment-989716573.

closed 2022-02-02 by matth-x