matth-x/MicroOcpp #34: FirmwareService.cpp warning

ChaitanyaYeole02, 2021-11-16:
@matth-x 
I am getting a warning when I am trying to compile my code on ArduinoIDE. 
Also, 
`HTTP_UPDATE_FAILED Error (-11): HTTP error: read Timeout`
I am not sure it this error because of this warning.
```
warning: lambda capture initializers only available with -std=c++14 or -std=gnu++14
     fwService->setOnInstall([fwService = fwService] (String &location) {

```

ChaitanyaYeole02, 2021-11-16:
My update firmware link is of https:// is this an issue?

matth-x, 2022-02-02:
Hi @ChaitanyaYeole02, did you succeed in the meantime?

The warnings should be fixed by now. Have you checked if you can access the update URL with a web browser from the same network as the controller?

closed 2022-03-03 by matth-x