matth-x/MicroOcpp #36: Websocket Server connection issue

akashdk191, 2021-12-08:
I am unable to connect NodeMCU to the OCPP server hosted in cloud while able to connect in localhost with IP : **192.168.1.11:9055/v1.6/alpha**
The same OCPP cloud server can be connected from chrome extension **Websocket King Client**.
There is no SSL enabled on cloud

Sample URL :
ws://splendidwasp21.loca.lt/v1.6/alpha.

Node MCU main.cpp macro definitions for server

**#define OCPP_HOST "splendidwasp21.loca.lt"
#define OCPP_PORT '\0'
#define OCPP_URL "v1.6/alpha"**

Is the above assignment correct to connect to cloud server.
if not what is the right way to connect to cloud server. Is something needs to be added during initialisation?

matth-x, 2021-12-09:
Hi,

You could try

```cpp
#define OCPP_HOST "splendidwasp21.loca.lt"
#define OCPP_PORT 9055
#define OCPP_URL "ws://splendidwasp21.loca.lt/v1.6/alpha"
```

Otherwise, you can play a little bit around by exchanging the canonical names with the IP address or adding the port number to the `OCPP_URL` field. In many cases, the problem lies anywhere between the microcontroller and the server though.

The best way to debug it, is to start a test sketch with the [WebSockets library](https://github.com/Links2004/arduinoWebSockets) only. Markus's example sketches get you started very quickly. To mimic an OCPP connection, you have to change the `protocol` parameter from `"arduino"` into `"ocpp1.6"` and that's it, then you can figure out what is necessary to change for my library.

akashdk191, 2021-12-09:
Hello Matth
 Thanks for the response. I tried the same with Websockets library too. the results remained same. And as suggested , juggled the names, port value, domain , entering actual IP address instead of domain name.
Nothing really worked out from ESP8266. Will try once with ESP32 to connect.

akashdk191, 2021-12-24:
this issue is fixed as per above url suggestion

closed 2021-12-24 by akashdk191