matth-x/MicroOcpp #37: Unable to initialize: could not open configuration file

HackLink123, 2021-12-21:
[image]
Hi @matth-x  ,I upload the project in the nodemcuv2,now the serial printf the following msg:
`[Configuration] Unable to initialize: could not open configuration file
[Configuration] Loading default configurations file failed!
[Configuration<T>] Initialized config object. Key = ChargeProfileMaxStackLevel, value = 20
[Configuration<T>] Initialized config object. Key = OCPP_STATE_TRANSACTION_ID_CONNECTOR_0, value = -1
[Configuration<T>] Initialized config object. Key = OCPP_STATE_AVAILABILITY_CONNECTOR_0, value = 2
[Configuration<T>] Initialized config object. Key = OCPP_STATE_TRANSACTION_ID_CONNECTOR_1, value = -1
[Configuration<T>] Initialized config object. Key = OCPP_STATE_AVAILABILITY_CONNECTOR_1, value = 2
[Configuration<T>] Initialized config object. Key = MeterValueSampleInterval, value = 60
[Configuration<T>] Initialized config object. Key = MeterValuesSampledDataMaxLength, value = 4
[Configuration<T>] Initialized config object. Key = HeartbeatInterval, value = 86400
[Configuration<const char*>] Initialized config object. Key = BUILD_NUMBER, value = 12345789
[OcppClientSocket] Disconnected!
[OcppOperation] Send requirement: [2,"531531531","BootNotification",{"chargePointModel":"My Charging Station","chargePointVendor":"My company name"}]
[OcppOperation] Send requirement: [2,"531531531","BootNotification",{"chargePointModel":"My Charging Station","chargePointVendor":"My company name"}]
[OcppClientSocket] Disconnected!`
Please tell me how I can solve this problem, thanks.

matth-x, 2021-12-21:
Hi, the Configuration "error" is a misleading message from my side, sorry. It only means that the library cannot find a configuration file on the flash memory (which is what you basically expect at the first run sequence). I will make it more meaningful.

The connection problem is something you will have to analyze and debug on your end. The simplest way to start is to create a new sketch with solely the [WebSockets library](https://github.com/Links2004/arduinoWebSockets) and see what you have to do to get a raw connection (you only have to change the WebSocket protocol parameter into `ocpp1.6`). If you manage to get a running raw WebSocket, the OCPP connection will likely work as well.

matth-x, 2022-02-02:
The debug message should be more comprehensive (changed in 4a623e0f1e622badebaccfd65aa30da734a95819). For the WS connection, please refer to https://github.com/matth-x/ArduinoOcpp/issues/36#issuecomment-989716573.

closed 2022-02-02 by matth-x