matth-x/MicroOcpp #38: uniqueptr+json  ambiguous issue

akashdk191, 2022-01-01:
[image]

This error is appearing in development branch library.
Whenever a new json document is declared with uniqueptr, swap ambiguous error is appearing during compilation.
Need help in resolving the compilation error.

matth-x, 2022-01-01:
That was addressed in bblanchon/ArduinoJson#1678 and is already fixed. Please update the dependency to the latest commit of the JSON library. Probably it will be sufficient to delete all folders in `.pio/libdeps` and to compile again.

akashdk191, 2022-01-02:
Thanks Matthias. 
 I was updating the platform.ini in a wrong location(inside libdeps folder) than in main project folder. It's resolved

closed 2022-01-02 by akashdk191