matth-x/MicroOcpp #40: not able to complie the code

rccess, 2022-01-04:
Hello Matt

Getting the following error the compile time, can you please advise me something so i can do it fix

[image]

Thank You

rccess, 2022-01-05:
@matth-x do you have chance to look on it once 

Thank you in advance

matth-x, 2022-01-05:
The example `OneConnector-EVSE` has three files and all must go into the source folder in order to compile.

rccess, 2022-01-06:
[image]

here is the folder structure inside the library 

but having same issue

[image]

matth-x, 2022-01-06:
They must go into the same folder as `sketch_jan04a.ino`

rccess, 2022-01-06:
thanks for your reply 

i have done the same but issue still same

[image]

[image]

rccess, 2022-01-06:
Matt,
can you please send any screenshot of folder setup so i can compare the same