matth-x/MicroOcpp #42: OCPP request not coming to server

rccess, 2022-01-12:
Hello Matt,

I have successfully uploaded the code in nodemcu and wifi also connected properly but nodemcu not sending request for ocpp connection to server, proper url defined 

same ocpp working with other ev charging machine

[image]

can you please advise on the same ?

any troubleshooting hint, i have tested with other websocket request test but no luck

please help

thank you

rccess, 2022-01-14:
matt, have any chance to help me please

bnystrom, 2022-01-18:
I believe OCPP_HOST should just be "my.server.com". In the past it's worked w/o the "ws://" 
OCPP_PORT 8010
OCPP_URL "/OCPP16/<chargeboxId>" (whatever the chargebox for your EVSE is)

Hope that helps some.

matth-x, 2022-02-02:
Hi @rccess, did you succeed with the input from @bnystrom?

Please refer to https://github.com/matth-x/ArduinoOcpp/issues/36#issuecomment-989716573 which describes what to do to get the initial connection right.

closed 2022-03-03 by matth-x