matth-x/MicroOcpp #43: Errors when building SECC example

ZolikovaPlesataHlava, 2022-01-14:
Hi.

I am having issues compiling the SECC example you provided as the try out sketch. I tried with arduino IDE and with platformIO. And I seem to get the same error on both builds. 

**Arduino IDE:** 
WARNING: library ArduinoOcpp-master claims to run on espressif8266, espressif32 architecture(s) and may be incompatible with your current board which runs on esp8266 architecture(s).
C:\Users\Matus\AppData\Local\Temp\arduino_modified_sketch_160544\main.ino: In function 'void setup()':
main:193:101: error: no matching function for call to 'WebSocketsClient::beginSSL(String&, uint16_t&, String&, const char [1], const char [8])'
             wSock->beginSSL(ocppUrlParsed.host, ocppUrlParsed.port, ocppUrlParsed.url, "", "ocpp1.6");
                                                                                                     ^
C:\Users\Matus\AppData\Local\Temp\arduino_modified_sketch_160544\main.ino:193:101: note: candidate is:
In file included from C:\Users\Matus\Documents\Arduino\libraries\ArduinoOcpp-master\src/ArduinoOcpp/Core/OcppSocket.h:29:0,
                 from C:\Users\Matus\Documents\Arduino\libraries\ArduinoOcpp-master\src/ArduinoOcpp/Core/OcppOperation.h:16,
                 from C:\Users\Matus\Documents\Arduino\libraries\ArduinoOcpp-master\src/ArduinoOcpp.h:8,
                 from C:\Users\Matus\AppData\Local\Temp\arduino_modified_sketch_160544\main.ino:20:
C:\Users\Matus\Documents\Arduino\libraries\arduinoWebSockets-master\src/WebSocketsClient.h:50:10: note: void WebSocketsClient::beginSSL(const char*, uint16_t, const char*, const uint8_t*, const char*)
     void beginSSL(const char * host, uint16_t port, const char * url = "/", const uint8_t * fingerprint = NULL, const char * protocol = "arduino");
          ^
C:\Users\Matus\Documents\Arduino\libraries\arduinoWebSockets-master\src/WebSocketsClient.h:50:10: note:   no known conversion for argument 1 from 'String' to 'const char*'
exit status 1
no matching function for call to 'WebSocketsClient::beginSSL(String&, uint16_t&, String&, const char [1], const char [8])'
**PlatformIO :**
> Executing task in folder OCPP4: C:\Users\Matus\.platformio\penv\Scripts\platformio.exe run --environment nodemcuv2 <

Warning! Ignore unknown configuration option `build_partitions` in section [env:esp32-development-board]
Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (3.2.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.30002.0 (3.0.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa 2.100300.210717 (10.3.0)
Converting main.ino
C:/Users/Matus/Documents/PlatformIO/Projects/OCPP4/src/main.ino:34: warning: "EV_PLUGGED" redefined
   34 | #define EV_PLUGGED LOW
      | 
C:/Users/Matus/Documents/PlatformIO/Projects/OCPP4/src/main.ino:31: note: this is the location of the previous definition
   31 | #define EV_PLUGGED HIGH
      | 
C:/Users/Matus/Documents/PlatformIO/Projects/OCPP4/src/main.ino:35: warning: "EV_UNPLUGGED" redefined
   35 | #define EV_UNPLUGGED HIGH
      | 
C:/Users/Matus/Documents/PlatformIO/Projects/OCPP4/src/main.ino:32: note: this is the location of the previous definition
   32 | #define EV_UNPLUGGED LOW
      |
C:/Users/Matus/Documents/PlatformIO/Projects/OCPP4/src/main.ino:47: warning: "EV_CHARGING" redefined
   47 | #define EV_CHARGING HIGH
      |
C:/Users/Matus/Documents/PlatformIO/Projects/OCPP4/src/main.ino:44: note: this is the location of the previous definition
   44 | #define EV_CHARGING LOW
      |
C:/Users/Matus/Documents/PlatformIO/Projects/OCPP4/src/main.ino:48: warning: "EV_SUSPENDED" redefined
   48 | #define EV_SUSPENDED LOW
      |
C:/Users/Matus/Documents/PlatformIO/Projects/OCPP4/src/main.ino:45: note: this is the location of the previous definition
   45 | #define EV_SUSPENDED HIGH
      | 
C:/Users/Matus/Documents/PlatformIO/Projects/OCPP4/src/main.ino:73: warning: "CHARGE_PERMISSION_ON" redefined
   73 | #define CHARGE_PERMISSION_ON LOW
      |
C:/Users/Matus/Documents/PlatformIO/Projects/OCPP4/src/main.ino:70: note: this is the location of the previous definition
   70 | #define CHARGE_PERMISSION_ON HIGH
      |
C:/Users/Matus/Documents/PlatformIO/Projects/OCPP4/src/main.ino:74: warning: "CHARGE_PERMISSION_OFF" redefined
   74 | #define CHARGE_PERMISSION_OFF HIGH
      |
[... 129 lines cut ...]
.pio\libdeps\nodemcuv2\WebSockets\src\WebSocketsServer.cpp: In constructor 'WebSocketsServer::WebSocketsServer(uint16_t, String, String)':
.pio\libdeps\nodemcuv2\WebSockets\src\WebSocketsServer.cpp:49:83: warning: 'void* memset(void*, int, size_t)' clearing an object of type 'struct WSclient_t' with no trivial copy-assignment; use assignment or value-initialization instead [-Wclass-memaccess]
   49 |     memset(&_clients[0], 0x00, (sizeof(WSclient_t) * WEBSOCKETS_SERVER_CLIENT_MAX));
      |                                                                                   ^
In file included from .pio\libdeps\nodemcuv2\WebSockets\src\WebSocketsServer.cpp:25:
.pio\libdeps\nodemcuv2\WebSockets\src\WebSockets.h:237:16: note: 'struct WSclient_t' declared here
  237 | typedef struct {
      |                ^
.pio\libdeps\nodemcuv2\WebSockets\src\WebSocketsServer.cpp: In member function 'bool WebSocketsServer::newClient(WiFiClient*)':      
.pio\libdeps\nodemcuv2\WebSockets\src\WebSocketsServer.cpp:465:23: warning: variable 'ip' set but not used [-Wunused-but-set-variable]
  465 |             IPAddress ip = client->tcp->remoteIP();
      |                       ^~
.pio\libdeps\nodemcuv2\WebSockets\src\WebSocketsServer.cpp: In member function 'void WebSocketsServer::handleNewClients()':
.pio\libdeps\nodemcuv2\WebSockets\src\WebSocketsServer.cpp:638:23: warning: variable 'ip' set but not used [-Wunused-but-set-variable]
  638 |             IPAddress ip = tcpClient->remoteIP();
      |                       ^~
*** [.pio\build\nodemcuv2\libf9e\WebSockets\WebSocketsClient.cpp.o] Error 1
==================================================== [FAILED] Took 5.44 seconds ====================================================

Environment    Status    Duration
-------------  --------  ------------
nodemcuv2      FAILED    00:00:05.445
=============================================== 1 failed, 0 succeeded in 00:00:05.445 ===============================================

I currently dont own any EVSE and I would love to try out OCPP to be used with my own backend system.

matth-x, 2022-02-02:
Hi @ZolikovaPlesataHlava,

In the previous `platformio.ini` file, not all dependencies were fully specified which led to compilation problems on some machines. Can you please load the most recent version of the library and the `platformio.ini` and try it again?

That's how the LDF output looks like on my machine:

```
Processing nodemcuv2 (platform: espressif8266@2.6.3; board: nodemcuv2; framework: arduino)
[...]
PLATFORM: Espressif 8266 (2.6.3) > NodeMCU 1.0 (ESP-12E Module)
[...]
Dependency Graph
|-- <ArduinoJson> 6.19.1
|-- <WebSockets> 2.3.6
|   |-- <ESP8266WiFi> 1.0        
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Hash> 1.0
|-- <LittleFS(esp8266)> 0.1.0    
|-- <ESP8266httpUpdate> 1.3      
|   |-- <ESP8266HTTPClient> 1.2  
|   |   |-- <ESP8266WiFi> 1.0    
|   |-- <ESP8266WiFi> 1.0        
|-- <ESP8266WiFi> 1.0
Building in release mode
[...]
```

ZolikovaPlesataHlava, 2022-02-07:
Perfect, the new platformio.ini fixed the problem.
Thank you.

closed 2022-02-07 by ZolikovaPlesataHlava

susanthomasv, 2022-03-05:
Which is the new platformio.ini? I'm getting the same errors

matth-x, 2022-03-07:
Can you try this? https://github.com/matth-x/ArduinoOcpp/blob/develop/platformio.ini

Please remember that you have to instruct PIO to download the new versions and replace the old ones.