matth-x/MicroOcpp #45: Warning compiling with VSCode and PlatformIO

shardul10, 2022-01-25:
I am using PlatformIO on MAC with VSCode. I have successfully compiled and run the code on ESP32. Thank you so much for this work. I do see two warnings that require higher version of c++ compiler. I am not sure what version the VScode compiler provided by Microsoft is at or how to upgrade. I wanted to know if these warning are serious and whether they will impact code at runtime. I am not familiar with lambda capture initializers. 

.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/ClearChargingProfile.cpp: In member function 'virtual void ArduinoOcpp::Ocpp16::ClearChargingProfile::processReq(ArduinoJson::JsonObject)':
.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/ClearChargingProfile.cpp:23:78: warning: lambda capture initializers only available with -std=c++14 or -std=gnu++14
     std::function<bool(int, int, ChargingProfilePurposeType, int)> filter = [payload = payload]

.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Tasks/FirmwareManagement/FirmwareService.cpp: In function 'ArduinoOcpp::FirmwareService* ArduinoOcpp::EspWiFi::makeFirmwareService(const char*)':
.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Tasks/FirmwareManagement/FirmwareService.cpp:317:30: warning: lambda capture initializers only available with -std=c++14 or -std=gnu++14
     fwService->setOnInstall([fwService = fwService] (String &location) {

.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Tasks/FirmwareManagement/FirmwareService.cpp:348:46: warning: lambda capture initializers only available with -std=c++14 or -std=gnu++14
     fwService->setInstallationStatusSampler([fwService = fwService] () {

matth-x, 2022-01-26:
Hi @shardul10, thank you for the feedback!

You are right, that warning is something serious. I've already fixed the issue in the development branch and will push it to the master branch very soon. In my tests, the usage of C++14 features with the PIO standard compilation parameters never led to any problematic behavior though, so if that's also the case for you, you can probably just continue with your current code base in the meantime.

closed 2022-01-26 by matth-x

shardul10, 2022-01-26:
Thank you for quick response. I haven't got that far to see if this is causing a problem. I will continue to use the code base until you release the change.