matth-x/MicroOcpp #48: OCPP_initialize function not calling

rccess, 2022-02-05:
Hello @matth-x 

I have successfully upload the code in esp8266 and it is connected to wifi but 

OCPP_initialize not calling

no ocpp request coming on central server 

[image]

Please do advise 

Thank You

matth-x, 2022-02-07:
Hi @rccess, in https://github.com/matth-x/ArduinoOcpp/issues/36#issuecomment-989716573 you can find how to make the WebSocket connection work.

rccess, 2022-02-07:
Service request already come to server and opened websocket request but not
booting and not charging station showing in cms but with same settings it
was working in your old repo

Please do advise

matth-x, 2022-02-07:
Is the `OCPP_loop()` inside the `loop()` function and will it be called in every `loop()` execution?

If both is true, please compile again with the following debug flags:

https://github.com/matth-x/ArduinoOcpp/blob/e742bc760950731309c9c9ce1e4391a69ac2da82/platformio.ini#L21-L23

matth-x, 2022-05-05:
Closed due to inactivity

closed 2022-05-05 by matth-x