matth-x/MicroOcpp #50: Consecutive startTransaction requests are sent, even before getting the response within the timeout period.

niravbhandari2000, 2022-02-24:
The logs look something like this:
 
```
[OcppOperation] Sent requirement (success): [2,"531531536","StartTransaction",{"connectorId":1,"meterStart":0,"timestamp":"2022-02-24T10:01:39.000Z","idTag":"vckjvcJvjk337nKJCn92"}]
[OcppOperation] Sent requirement (success): [2,"531531536","StartTransaction",{"connectorId":1,"meterStart":0,"timestamp":"2022-02-24T10:01:39.000Z","idTag":"vckjvcJvjk337nKJCn92"}]
[OcppClientSocket] get pong
[OcppOperation] Sent requirement (success): [2,"531531536","StartTransaction",{"connectorId":1,"meterStart":0,"timestamp":"2022-02-24T10:01:39.000Z","idTag":"vckjvcJvjk337nKJCn92"}]
[OcppClientSocket] get pong
[OcppOperation] Sent requirement (success): [2,"531531536","StartTransaction",{"connectorId":1,"meterStart":0,"timestamp":"2022-02-24T10:01:39.000Z","idTag":"vckjvcJvjk337nKJCn92"}]
```

valerio-2020, 2022-02-24:
Happening with StopTransaction as well. The following are the console logs - 

```
[OcppOperation] Sent requirement (success): [2,"531531539","StopTransaction",{"meterStop":2,"timestamp":"2022-02-24T13:54:20.000Z","transactionId":58}]
[OcppOperation] Sent requirement (success): [2,"531531539","StopTransaction",{"meterStop":2,"timestamp":"2022-02-24T13:54:20.000Z","transactionId":-1}]
[OcppClientSocket] get pong
[OcppOperation] Sent requirement (success): [2,"531531539","StopTransaction",{"meterStop":2,"timestamp":"2022-02-24T13:54:20.000Z","transactionId":-1}]
[OcppClientSocket] get pong
[OcppOperation] Sent requirement (success): [2,"531531539","StopTransaction",{"meterStop":2,"timestamp":"2022-02-24T13:54:20.000Z","transactionId":-1}]
[OcppClientSocket] get pong
[OcppOperation] Sent requirement (success): [2,"531531539","StopTransaction",{"meterStop":2,"timestamp":"2022-02-24T13:54:20.000Z","transactionId":-1}]
```

valerio-2020, 2022-02-24:
One thing that I understand here is, whenever a request is being sent (in my case heartbeat) and a proper .conf() response is not coming back. The code retries until timeout. But if another request like Start/Stop transaction is made within that time period, it fails. 

Look at the following console logs - 

```
[OcppOperation] Sent requirement (success): [2,"531531540","Heartbeat",{}]
[OcppClientSocket] get pong
Trig HIGH **I am triggering the authorize function here**
EV_CHARGE_PIN set to high
[OcppOperation] Sent requirement (success): [2,"531531540","Heartbeat",{}]
[OcppClientSocket] get pong
[OcppOperation] Sent requirement (success): [2,"531531540","Heartbeat",{}]
[OcppEngine] Discarding operation due to timeout: [OcppOperation] I am a Authorize
[OcppOperation] Sent requirement (success): [2,"531531540","Heartbeat",{}]
[OcppClientSocket] get pong
```

@matth-x How to deal with this?

matth-x, 2022-02-28:
Hi @niravbhandari2000 @valerio-2020 

Thank you for the feedback! I admit that the console output is not 100% clear here. The message

```
[OcppOperation] Sent requirement (success): [...]
```

means that the OCPP library _assumes_ that the message was sent successfully. But it actually doesn't know and just waits for the response from the destination with a timeout and retry mechanism. That explains why one operation appears multiple times on the log.

The log in https://github.com/matth-x/ArduinoOcpp/issues/50#issue-1149183983 looks completely normal. The log in https://github.com/matth-x/ArduinoOcpp/issues/50#issuecomment-1049885314 actually shows a bug, because the library tries to resend with a different OCPP message (the `transactionId` differs). Thank you for the report! I will investigate it and come back to you. https://github.com/matth-x/ArduinoOcpp/issues/50#issuecomment-1049912159 shows that different operations can have different timeouts. A `Heartbeat` message waits longer when the EVSE is offline, however the `Authorize` message has a quick timeout, because the EV driver shouldn't be waiting for more than 30 seconds after presenting the NFC card in case the EVSE is offline. So the `Authorize` operation will just be discarded in the meantime. You can see how to control timeouts in `ArduinoOcpp.cpp`. If you need a custom timeout, just copy the contents of `authorize(...)` into your implementation and set the parameters according to your needs.

matth-x, 2022-03-01:
Checked it and the bug in https://github.com/matth-x/ArduinoOcpp/issues/50#issuecomment-1049912159 is already addressed in the develop branch

closed 2022-03-01 by matth-x