matth-x/MicroOcpp #51: The values sent in any response has to be a String according to OCPP documentation, but the library sends values as Integers or Float

valerio-2020, 2022-02-26:
The response for MeterValues looks like this - 

[OcppOperation] Sent requirement (success): [2,"531531590","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2022-02-25T13:20:29.000Z","sampledValue":[{"value":0,"measurand":"Power.Active.Import","unit":"W"}]}]}]

The value here should be a String, but it is an int value.

matth-x, 2022-02-28:
@valerio-2020 Thank you very much for the feedback! Actually I've already addressed this on the develop branch.

If you find further formatting problems, I will be happy if you post them here!

closed 2022-02-28 by matth-x

valerio-2020, 2022-03-01:
Hey @matth-x 
Thanks for the response. Can you tell me where the changes were made so that I can incorporate that in my code as well. I am currently using PlatformIO for your library. 
Thanks.

matth-x, 2022-03-01:
In PIO you can directly point to this GitHub repo and its develop branch. The `platformio.ini` would look like:

```
lib_deps =
    ; matth-x/ArduinoOcpp@0.0.4 <-- replace this
    https://github.com/matth-x/ArduinoOcpp.git#develop ; (always the most recent version, but it changes frequently)
```
or
```
lib_deps =
    ; matth-x/ArduinoOcpp@0.0.4 <-- replace this
    https://github.com/matth-x/ArduinoOcpp.git#902315381ccbccbe8a5673a8e87ebba5f9c8a239 ; (a very recent version, but it only changes if you update the hash)
```

Still, you must reload the code in PIO. You can find good instructions on Google.