matth-x/MicroOcpp #55: Use with Grizzl-E smart charger

tube0013, 2022-03-18:
Hi, the Grizzl-E smart charger uses an ESP32 for Wi-Fi and it's application. Their firmware is a mess, and not very compliant to OCPP. What all would be required to see if ArduinoOcpp could control this box?

This issue has some pictures and details on the fw provided.

https://github.com/lbbrhzn/ocpp/issues/442

It may be possible to follow the traces on the esp32 daughter board there is an 8 pin header that connects is to the main pcb. There is a buzzer and led so I'm guessing those would be 2 of the gpio pins. 

Appreciate any input
Thank you

matth-x, 2022-03-19:
Sounds like a great project! The first thing to find out is how the communication on their board works in detail and how to control the EVSE from an Arduino sketch. After that you can customize the [SECC sketch](https://github.com/matth-x/ArduinoOcpp/tree/master/examples/SECC) to support that communication mechanism. Looking forward to hearing about the progress!

matth-x, 2023-02-02:
Feel free to reopen this issue in case you retrofit a charger using this project.

closed 2023-02-02 by matth-x