matth-x/MicroOcpp #60: authorization key exchange

humanlyj, 2022-04-06:
Dear all,

I've using this library and it's good. Really appreciate.

I have a issue about that 'changeconfiguration'.
I got a new key from server by 'changeconfiguration' message for authorization.
But I couldn't find the code to make new authorization key.

Authorization key was not changed after receive the new key from server. Then my module fail to register because it was re-connect to server with same authorization key.

May I missed something? Please help me.

humanlyj, 2022-04-06:
Oh, one more question,

At the file 'ChangeConfiguration.cpp'-'processReq',
The authorization key from server is must digit value?

matth-x, 2023-01-07:
The [Mongoose WebSocket Adapter](https://github.com/matth-x/ArduinoOcppMongoose) for ArduinoOcpp supports the Authorization key exchange as defined in the OCPP standard now. But the included WebSockets integration still misses that function, sorry.

Please reopen for follow-up questions.

closed 2023-01-07 by matth-x