matth-x/MicroOcpp #61: fatal error: memory: No such file or directory  #include <memory>

PatrikForis, 2022-04-26:
Hello. I trying to compile code form examples of this project but I got fatal error message and I can´t move on. Can somebody please help with solution of this error?

[image]

marcos-mendez, 2022-05-02:
You are running the compilation on wrong settings in the Arduino IDE. Recommend closing the Issue.

PatrikForis, 2022-05-03:
Thank you for answer, but which setting should I change.

marcos-mendez, 2022-05-03:
I should check it personally but you have to check mainly board settings

matth-x, 2023-01-07:
Sorry, couldn't reproduce this.

Closed due to inactivity. Please reopen for follow-up questions.

closed 2023-01-07 by matth-x