matth-x/MicroOcpp #62: Is this library ESP32-S3-DevKitC-1-N8R8 compatible

vinothkrstatiq, 2022-05-13:
while compiling the board i got below error 
Arduino: 1.8.19 (Windows 10), Board: "ESP32S3 Dev Module, Disabled, QIO 80MHz, 8MB (64Mb), Core 1, Core 1, Hardware CDC and JTAG, Disabled, Disabled, Disabled, UART0 / Hardware CDC, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), 921600, None"

WARNING: library ArduinoOcpp claims to run on espressif8266, espressif32 architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).

C:\Users\Vinoth\Documents\Arduino\libraries\arduinoWebSockets-master\src\WebSocketsServer.cpp: In member function 'WSclient_t* WebSocketsServerCore::handleNewClient(WiFiClient*)':

C:\Users\Vinoth\Documents\Arduino\libraries\arduinoWebSockets-master\src\WebSocketsServer.cpp:634:12: warning: function may return address of local variable [-Wreturn-local-addr]

     return client;

            ^~~~~~

C:\Users\Vinoth\Documents\Arduino\libraries\arduinoWebSockets-master\src\WebSocketsServer.cpp:626:20: note: declared here

         WSclient_t dummy = WSclient_t();

                    ^~~~~

C:\Users\Vinoth\Documents\Arduino\libraries\LITTLEFS\src\esp_littlefs.c:19:2: warning: #warning ("Use the built-in LITTLEFS library") [-Wcpp]

 #warning("Use the built-in LITTLEFS library")

  ^~~~~~~

C:\Users\Vinoth\Documents\Arduino\libraries\LITTLEFS\src\LITTLEFS.cpp: In member function 'virtual bool LITTLEFSImpl::exists(const char*)':

C:\Users\Vinoth\Documents\Arduino\libraries\LITTLEFS\src\LITTLEFS.cpp:44:28: error: no matching function for call to 'LITTLEFSImpl::open(const char*&, const char [2])'

     File f = open(path, "r");

                            ^

In file included from C:\Users\Vinoth\Documents\Arduino\libraries\LITTLEFS\src\LITTLEFS.cpp:17:

C:\Users\Vinoth\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\FS\src/vfs_api.h:38:17: note: candidate: 'virtual fs::FileImplPtr VFSImpl::open(const char*, const char*, bool)'

     FileImplPtr open(const char* path, const char* mode, const bool create) override;

                 ^~~~

C:\Users\Vinoth\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\FS\src/vfs_api.h:38:17: note:   candidate expects 3 arguments, 2 provided

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\Vinoth\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi

 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

Multiple libraries were found for "SD.h"

 Used: C:\Users\Vinoth\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\SD

 Not used: C:\Program Files (x86)\Arduino\libraries\SD

Multiple libraries were found for "WebSocketsClient.h"

 Used: C:\Users\Vinoth\Documents\Arduino\libraries\arduinoWebSockets-master

 Not used: C:\Users\Vinoth\Documents\Arduino\libraries\WebSockets

exit status 1

Error compiling for board ESP32S3 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

matth-x, 2022-05-19:
Hi @vinothkrstatiq 

Thank you for the report. The compiler output shows that the library versions are not compatible with each other. Does it also print error messages if you compile a blank sketch with only arduinoWebSockets? What if you specify the same version numbers as in the [platformio.ini](https://github.com/matth-x/ArduinoOcpp/blob/master/platformio.ini)?

I see that the ESP32-S3 support has just been released (https://github.com/espressif/arduino-esp32/issues/5594). I will also do checks for my library and notify you if can reproduce this error.

matth-x, 2023-08-25:
I'm closing this assuming it's no longer relevant.

closed 2023-08-25 by matth-x