matth-x/MicroOcpp #66: EVSE state jumps from Available to Charging even before startTransaction

whuizdis, 2022-06-08:
Hello @matth-x , Continuing to make use of this and trying to adapt on my application with ESP32.

I have been seeing the state transition different in 2 different ESP32's I have (1 - ESP32-WROOM-32DC not the dev board) and the other 2 - ESP-WROOM-32 development board)

With #1, EVSE state is jumping from Available to Charging even without any startTransaction command and with the exact same Firmware (basically same as that of SECC example), on #2, Its behaving as expected and transitioning to Preparing state after Available. What could be the culprit here? Please help!

On #1, here are my logs:

00:10:17.505 -> [AO] info (/Users/phanigalla/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:103): init new configurations container: /volatile
00:10:19.472 -> [AO] info (/Users/phanigalla/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:34): Connected to url: ws://xxxxPhaniTest/
00:10:19.542 -> [AO] info (/Users/phanigalla/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/BootNotification.cpp:94): Request has been accepted
00:10:19.542 -> [AO] info (/Users/phanigalla/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:48): New status: Available
00:10:19.575 -> [AO] info (/Users/phanigalla/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:48): New status: Charging
00:10:19.575 -> [AO] info (/Users/phanigalla/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Tasks/SmartCharging/SmartChargingService.cpp:80): New limit for connector 1, scheduled at = 2022-06-08T18:40:19.000Z, nextChange = 2037-01-01T00:00:00.000Z, limit = 11000.000000
00:10:19.611 -> [main] Smart Charging allows maximum charge rate: 11000.00

Whereas on #2 (With development board) its going to Preparing state as per expectation.

00:14:48.434 -> [AO] info (/Users/phanigalla/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:103): init new configurations container: /volatile
00:14:50.579 -> [AO] info (/Users/phanigalla/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:34): Connected to url: ws://xxxxPhaniTest/
00:14:50.651 -> [AO] info (/Users/phanigalla/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/BootNotification.cpp:94): Request has been accepted
00:14:50.651 -> [AO] info (/Users/phanigalla/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:48): New status: Available
00:14:50.651 -> [AO] info (/Users/phanigalla/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:48): New status: Preparing
00:14:50.686 -> [AO] info (/Users/phanigalla/Documents/Arduino/libraries/ArduinoOcpp/src/ArduinoOcpp/Tasks/SmartCharging/SmartChargingService.cpp:80): New limit for connector 1, scheduled at = 2010-01-01T00:00:00.000Z, nextChange = 2037-01-01T00:00:00.000Z, limit = 11000.000000
00:14:50.686 -> [main] Smart Charging allows maximum charge rate: 11000.00

matth-x, 2022-06-09:
Hi @whuizdis 

Thank you for the report! You see two status messages because OCPP sends one for the connector status and the other one for the whole charger. I'm really sorry about the misleading debug message. I have made it more precise now, but it will take some time until the change makes it into the newest release.

In the meantime, you can set the build flag `AO_TRAFFIC_OUT` which will show the final OCPP messages on the monitor. Hopefully that will help you with the further development.

bnystrom, 2022-06-09:
Status about the "entire" EVSE are generally sent as "Connector  0". You normally only see this for EVSEs that have multiple connectors, but only one connector can be utilized at a time ( In the case of a DC Fast charger that has both a CCS and CHAdeMO connector). You normally (in my experience) don't see a Status Notification Connector 0 for AC L2 EVSEs with one or two connectors. Each connector in those cases is independent of the other, so ne need for a Connector 0 status.

matth-x, 2022-06-10:
@bnystrom Thank you for your input. I will consider to add that as a default option since I'm aware now that the status messages for connectorId 0 leads to confusion.

By chance, have you observed further uncommon traits of this library?

whuizdis, 2022-06-11:
Thanks for your help @matth-x  and for that suggestion @bnystrom . I made those changes. 

Have an unrelated question just posting here for continuation and need your help on these 2 questions. 

I'm expecting MeterValues message to only get sent once the Transaction is initiated and Stopped one the transaction is stopped, But I'm seeing MeterValues sent always. 

1. Is there anything I can do to get it align with Start and Stop transactions only?
2. Also, Whenever Transaction is started, MeterValues() message should be sent with meterStart readings and meterStop readings for Energy. At the moment, I can see the MeterValues message only get sent at its own frequency. How do I get this sorted?

bnystrom, 2022-06-12:
I would agree. MeterValues should only be transmitted after a valid
StartTransaction, and stop being transmitted at the time of a
StopTransaction. MeterValues require a valid TransactionID, and that only
exists while a transaction is occurring.

That being said, there are implementations where the EVSE will send a
meterValue for connectorID 0 as a running metric of the entire EVSE, but
it's not associated with any particular transaction. (I'm not a fan of the
connector 0 stuff myself, but I'm sure it has uses)

The first and last MeterValue energy values do not need to contain a
matching meterStart and meterStop value that is sent in the Start and Stop
transaction messages. And different implementations interpret those values
differently at times. Some implementations always use a zero based
meterStart, while others use an incremental one.  So, it is possible that
you will get a higher "meterStart" value in the StartTransation than the
initial energy value in the meterValue message. That indicates that
meterStart is incremental over several transactions, while the meterValue
energy is zero based.

Also, depending on the "MeterValueSampleInterval", the first meterValue may
not be sent from the EVSE for several minutes after the transaction
actually starts, and the last meterValue may be sent several minutes prior
to the end of a transaction. I guess what I'm saying here is that
meterValues are not required to be sent the instant a transaction starts
and stops. Many implementations delay the sending of the first meterValue
until "MeterValueSampleInterval" time.

Sorry to say, but the OCPP1.x specification leaves a lot to open
interpretation. The OCA committee working on OCPP 2.x is working hard at
closing some of those loose interpretations.

Best Regards
Bryan Nystrom

whuizdis, 2022-06-14:
Thank you @bnystrom for your views. I also believe the better way of sending MeterValues is only when we are in transaction. Also, I think with this library, @matth-x added meterStart and meterStop values (which can be Non Zero). 

For the first part, I'm wondering how do I control (where to change in this library) the ability to have meterValues triggered/stopped with transaction start/stop. I wanted to at least have it as a configurable.

matth-x, 2022-06-18:
@bnystrom Thank you for continuing the discussion. Your input is highly appreciated.

Meter Values are always associated with the respective transactionId if they are taken during a transaction. Otherwise they aren't assigned to a transaction and the transactionId field is omitted.

I will make it the default behavior to not send periodic Meter Values when no transaction is ongoing. Thank you for your feedback here!

The zero-based meter readings are no internal feature of this library yet, though I'm considering it too (there has been more feedback about it). Today, this can be achieved by passing a zero-based energy reader to the library instead of a monotonous energy meter. The library doesn't do any internal checks on meter readings (you could also count backwards during a transaction if the charging system integrates photovoltaics for example).

@whuizdis StartTransaction and StopTransaction use the same energy measurand as MeterValues (in the default configuration). You can collect all values into a single data structure and that way you will get a full, continuous view on the transactions.

whuizdis, 2022-07-13:
Thanks @matth-x, Yes I could save all the Energy Values and they are good. Only question is to get the MeterValues only sent when we are in a transaction. Can you help me with changes to get the MeterValues triggered only when we are in transaction?

matth-x, 2022-07-14:
You could filter the Meter Values. The `transactionId` field of a Meter Values message assigns it to a specific transaction. If no `transactionId` field is present, the Meter Value does not belong to any transaction and you could remove it from your data set.

It's already on my to-do list to not capture Meter Values outside of transactions by default but I would recommend to implement the solution above in the meantime.

matth-x, 2022-11-13:
@bnystrom FYI, 03177f93e7d2c6b6b15c07b5d0f9944da65512b0 addresses sending MeterValues outside of transactions. Now it's possible to control it with the configuration `AO_MeterValuesInTxOnly` (true by default). Thank you again for the input.