matth-x/MicroOcpp #69: Not Connecting with OCPP server

VjSargunam, 2022-07-16:
Hi Matt,
         As per the instruction i installed the ArduinoOcpp in to ESP32. The project got successfully built. But after uploading the code to esp32 it is not connected with the central server it give be warnings as shown below in the serial terminal window.
[main] Wait for WiFi: ... connected!
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:103): init new configurations container on flash filesystem: /volatile

Can you please explain me what is the above message is all about,

Here is my code

```
#include <Arduino.h>
#include <WiFiClientSecure.h>
#include <HTTPClient.h>
#include <Update.h>
#include <FS.h>
#include <ArduinoOcpp.h>

#define STASSID "username"
#define STAPSK  "password*"

#define OCPP_HOST "echo.websocket.events" // I added my own hot address here
#define OCPP_PORT 80
#define OCPP_URL "ws://echo.websocket.events/"

void setup() {
    Serial.begin(9600);
    Serial.setDebugOutput(true);
    Serial.print(F("[main] Wait for WiFi: "));

    WiFi.begin(STASSID, STAPSK);
    while (!WiFi.isConnected()) {
        Serial.print('.');
        delay(1000);
    }

    Serial.print(F(" connected!\n"));

    //Initialize the OCPP library

    OCPP_initialize(OCPP_HOST, OCPP_PORT, OCPP_URL);

    
    //Notify the Central System that this station is ready
     
    bootNotification("DC-001", "Company name", [] (JsonObject confMsg) {
        //This callback is executed when the .conf() response from the central system arrives
          Serial.print(F("BootNotification was answered. Central System clock: "));
          Serial.println(confMsg["currentTime"].as<String>()); //"currentTime" is a field of the central system response
        
        //Notify your hardare that the BootNotification.conf() has arrived. E.g.:
        //evseIsBooted = true;
    });
    

  
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

And here is the compilation messages in the terminal
 
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- WiFi @ 2.0.0
|-- LittleFS_esp32 @ 1.0.5+sha.11d47df
|   |-- FS @ 2.0.0
|-- HTTPUpdate @ 2.0.0
|   |-- HTTPClient @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- ArduinoOcpp @ 0.1.0
|   |-- ArduinoJson @ 6.19.1
|   |-- WebSockets @ 2.3.6
|   |   |-- Ethernet @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |-- LittleFS_esp32 @ 1.0.5+sha.11d47df
|   |   |-- FS @ 2.0.0
|   |-- HTTPUpdate @ 2.0.0
|   |   |-- HTTPClient @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |   |-- WiFi @ 2.0.0
|   |   |-- Update @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- FS @ 2.0.0
|-- FS @ 2.0.0
|-- HTTPClient @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- Update @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|   |-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Linking .pio\build\esp32dev\firmware.elf
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.3% (used 40208 bytes from 327680 bytes)
Flash: [========  ]  78.0% (used 1021893 bytes from 1310720 bytes)
Building .pio\build\esp32dev\firmware.bin
esptool.py v3.3
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
=================================== [SUCCESS] Took 296.42 seconds ===================================
 *  Terminal will be reused by tasks, press any key to close it.

rathourcode, 2023-05-05:
> Hi Matt, As per the instruction i installed the ArduinoOcpp in to ESP32. The project got successfully built. But after uploading the code to esp32 it is not connected with the central server it give [...]

were you able to connect to ocppp server

matth-x, 2023-08-25:
I'm closing this assuming it's no longer relevant.

closed 2023-08-25 by matth-x