matth-x/MicroOcpp #71: Getting error after upgrading to 0.1.0 (from 0.0.4)

shardul10, 2022-07-23:
I recently upgraded to 0.1.0. 

I am getting this compilation error:
src/OneConnector_HW_integration.cpp: In function 'void setup()':
src/OneConnector_HW_integration.cpp:558:47: error: no matching function for call to 'OCPP_initialize(ArduinoOcpp::EspWiFi::OcppClientSocket*&, float, ArduinoOcpp::FilesystemOpt::Mode, ArduinoOcpp::OcppClock&)'
             ArduinoOcpp::Clocks::DEFAULT_CLOCK);

For the code:
  OCPP_initialize(oSock,
            /* Grid voltage */ 230.f,
            ArduinoOcpp::FilesystemOpt::Use_Mount_FormatOnFail,
            ArduinoOcpp::Clocks::DEFAULT_CLOCK);

This code part is from the example code SECC. 

I also see startTransaction now takes idTag as first argument. Is there any place where I can see changes that have happened between the versions, especially the ones not backward compatible.

stig888881, 2022-07-24:
Hello, what error does it give you?

shardul10, 2022-07-24:
Well, the startTransaction() now requires const char *idTag as first argument. In my old code (with 0.0.4), this function call did not require this argument. Just wondering if there any other changes documented that I should be aware of.

stig888881, 2022-07-24:
> 
have you tried running the example ? it doesn't work for me. The server gives an error, there is no IdTagInfo field in the transaction request.

matth-x, 2022-07-24:
@shardul10 In version 0.0.4, `startTransaction(...)` had no dedicated idTag parameter but used the cached idTag from a previous Authorize operation. But in case you authorize your idTags locally, that leads to a superfluous message and after all, it didn't really simplify the transaction process. Apart from that and refactoring some data types, I didn't change the functionality of the existing functions.

@stig888881 Can you try capturing the console output when triggering a StartTransaction? Would be helpful, thanks!

shardul10, 2022-07-25:
Thanks @matth-x. Just wanted to be sure there is no other changes. I did change my code to pass idTag to startTransaction and it works.

shardul10, 2022-07-25:
> have you tried running the example ? it doesn't work for me. The server gives an error, there is no IdTagInfo field in the transaction request.

Yes, the example works.

matth-x, 2023-08-25:
Closing this due to inactivity

closed 2023-08-25 by matth-x