matth-x/MicroOcpp #72: Testing with a button

stig888881, 2022-07-23:
I'm trying this code, but the server reports that idTagInfoid_tag_info is missing
`// matth-x/ArduinoOcpp
// Copyright Matthias Akstaller 2019 - 2022
// MIT License

#include <Arduino.h>
#if defined(ESP8266)
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
ESP8266WiFiMulti WiFiMulti;
#elif defined(ESP32)
#include <WiFi.h>
#else
#error only ESP32 or ESP8266 supported at the moment
#endif

#include <ArduinoOcpp.h>

#define STASSID "Keenetic-5742"
#define STAPSK  "88888888"

#define OCPP_HOST "192.168.1.34"
#define OCPP_PORT 9000
#define OCPP_URL "ws://192.168.1.34/"
#define SS_PIN 4  //D2
#define RST_PIN 5 //D1

#include <SPI.h>
#include <MFRC522.h>

MFRC522 mfrc522(SS_PIN, RST_PIN);   // Create MFRC522 instance.
int statuss = 0;
int out = 0;
int button = 15; //D8(gpio15)
int buttonState=0;
char *idTag;

//
////  Settings which worked for my SteVe instance
//
//#define OCPP_HOST "my.instance.com"
//#define OCPP_PORT 80
//#define OCPP_URL "ws://my.instance.com/steve/websocket/CentralSystemService/gpio-based-charger"

void setup() {
    pinMode(button, INPUT);
  SPI.begin();      // Initiate  SPI bus
  mfrc522.PCD_Init();   // Initiate MFRC522
    /*
     * Initialize Serial and WiFi
     */ 

    Serial.begin(115200);
    Serial.setDebugOutput(true);

    Serial.print(F("[main] Wait for WiFi: "));

#if defined(ESP8266)
    WiFiMulti.addAP(STASSID, STAPSK);
    while (WiFiMulti.run() != WL_CONNECTED) {
        Serial.print('.');
        delay(1000);
    }
#elif defined(ESP32)
    WiFi.begin(STASSID, STAPSK);
    while (!WiFi.isConnected()) {
        Serial.print('.');
        delay(1000);
    }
#else
#error only ESP32 or ESP8266 supported at the moment
#endif

    Serial.print(F(" connected!\n"));

    /*
     * Initialize the OCPP library
     */
    OCPP_initialize(OCPP_HOST, OCPP_PORT, OCPP_URL);

    /*
     * Integrate OCPP functionality. You can leave out the following part if your EVSE doesn't need it.
     */
    setPowerActiveImportSampler([]() {
        //measure the input power of the EVSE here and return the value in Watts
        return 0.f;
    });

    setOnChargingRateLimitChange([](float limit) {
        //set the SAE J1772 Control Pilot value here
        Serial.print(F("[main] Smart Charging allows maximum charge rate: "));
        Serial.println(limit);
    });

    setEvRequestsEnergySampler([]() {
        //return true if the EV is in state "Ready for charging" (see https://en.wikipedia.org/wiki/SAE_J1772#Control_Pilot)
        return false;
    });

    //... see ArduinoOcpp.h for more settings

    /*
     * Notify the Central System that this station is ready
     */
    bootNotification("My Charging Station", "My company name");
    
}

void loop() {
    buttonState=digitalRead(button);
     if (buttonState == 1)
 {
Serial.print("PUSH");
 delay(200);
 }
    /*
     * Do all OCPP stuff (process WebSocket input, send recorded meter values to Central System, etc.)
     */
    OCPP_loop();
    /*
     * Check internal OCPP state and bind EVSE hardware to it
     */

    /*
     * Detect if something physical happened at your EVSE and trigger the corresponding OCPP messages
     */
    if ( mfrc522.PICC_IsNewCardPresent()) {
      mfrc522.PICC_ReadCardSerial();
       String content= "";
        //const char *idTag = "0123456789abcd"; //e.g. idTag = RFID.readIdTag();
         for (byte i = 0; i < mfrc522.uid.size; i++) 
  {
     Serial.print(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " ");
     Serial.print(mfrc522.uid.uidByte[i], HEX);
     content.concat(String(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " "));
     content.concat(String(mfrc522.uid.uidByte[i], HEX));
  }
    content.toUpperCase();
        idTag=strcpy(new char[content.length()+1],content.c_str());
        authorize(idTag, [] (JsonObject response) {
            //check if user with idTag is authorized
            if (!strcmp("Accepted", response["idTagInfo"]["status"] | "Invalid")){
                Serial.println(F("[main] User is authorized to start charging"));
            } else {
                Serial.printf("[main] Authorize denied. Reason: %s\n", response["idTagInfo"]["status"] | "");
            }
        });
        Serial.printf("[main] Authorizing user with idTag %s\n", idTag);
    }

    
    if (buttonState == 1) {//e.g. authorized idTag from above
        startTransaction(idTag, [] (JsonObject response) {
            //Callback: Central System has answered. Could flash a confirmation light here.
            Serial.printf("[main] Started OCPP transaction. Status: %s, transactionId: %u\n",
                    response["idTagInfo"]["status"] | "Invalid",
                    response["transactionId"] | -1);
        });
    }
    
    if (false) {
        stopTransaction([] (JsonObject response) {
            //Callback: Central System has answered.
            Serial.print(F("[main] Stopped OCPP transaction\n"));
        });
    }

    //... see ArduinoOcpp.h for more possibilities
}`

matth-x, 2022-07-24:
Can you print the idTag before calling `authorize` and `startTransaction` and post the whole console output?

stig888881, 2022-07-25:
> Can you print the idTag before calling `authorize` and `startTransaction` and post the whole console output?

Here's what the client sends:
connected!
[AO] info (Configuration.cpp:103): init new configurations container: /volatile
[AO] info (OcppSocket.cpp:34): Connected to url: ws://192.168.1.34/
[AO] To WS lib: [2,"1000000","BootNotification",{"chargePointModel":"My Charging Station","chargePointVendor":"My company name"}]
[AO] From WS lib: [3,"1000000",{"currentTime":"2022-07-25T15:48:56.748659","interval":10,"status":"Accepted"}]
[AO] info (BootNotification.cpp:94): Request has been accepted
[AO] From WS lib: WS pong
[AO] info (StatusNotification.cpp:48): New status: Available
[AO] info (StatusNotification.cpp:48): New status: Preparing
[AO] info (SmartChargingService.cpp:80): New limit for connector 1, scheduled at = 2022-07-25T15:48:56.000Z, nextChange = 2037-01-01T00:00:00.000Z, limit = 11000.000000
[main] Smart Charging allows maximum charge rate: 11000.00
[AO] To WS lib: [2,"1000001","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2022-07-25T15:48:56.000Z"}]
[AO] From WS lib: [3,"1000001",{}]
[AO] To WS lib: [2,"1000002","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2022-07-25T15:48:56.000Z"}]
[AO] From WS lib: [3,"1000002",{}]
pm open,type:2 0
[AO] To WS lib: [2,"1000003","Heartbeat",{}]
[AO] From WS lib: [3,"1000003",{"currentTime":"2022-07-25T15:49:02.225972"}]
[main] Authorizing user with idTag  F6 CA 0C C5
[AO] To WS lib: [2,"1000005","Authorize",{"idTag":" F6 CA 0C C5"}]
[AO] From WS lib: [3,"1000005",{"idTagInfo":{"status":"Accepted"}}]
[AO] info (Authorize.cpp:41): Request has been accepted
[main] User is authorized to start charging

I press the button and:
[AO] warning (StartTransaction.cpp:60): Started transaction while OCPP already presumes a running transaction
[AO] info (StartTransaction.cpp:66): StartTransaction initiated
[AO] To WS lib: [2,"1000040","StartTransaction",{"connectorId":1,"meterStart":0,"timestamp":"2022-07-25T15:49:24.000Z","idTag":" F6 CA 0C C5"}]
[AO] From WS lib: [4,"1000040","InternalError","An unexpected error occurred.",{}]

Here's what the server responds to:
 receive message [2,"1000040","StartTransaction",{"connectorId":1,"meterStart":0,"timestamp":"2022-07-25T15:49:24.000Z","idTag":" F6 CA 0C C5"}]
ERROR:ocpp:Error while handling request '<Call - unique_id=1000040, action=StartTransaction, payload={'connectorId': 1, 'meterStart': 0, 'timestamp': '2022-07-25T15:49:24.000Z', 'idTag': ' F6 CA 0C C5'}>'
Traceback (most recent call last):
  File "C:\Users\Dimas\AppData\Roaming\Python\Python38\site-packages\ocpp\charge_point.py", line 194, in _handle_call
    response = handler(**snake_case_payload)
  File "C:\Users\Dimas\AppData\Roaming\Python\Python38\site-packages\ocpp\routing.py", line 47, in inner
    return func(*args, **kwargs)
  File "C:/Users/Dimas/PycharmProjects/OCPP_Server1.6/main.py", line 48, in on_start_transaction
    return call_result.StartTransactionPayload(
TypeError: __init__() missing 1 required positional argument: 'id_tag_info'
INFO:ocpp:ws://192.168.1.34: send [4,"1000040","InternalError","An unexpected error occurred.",{}]

stig888881, 2022-07-25:
> Can you print the idTag before calling `authorize` and `startTransaction` and post the whole console output?

I really hope for your help.

matth-x, 2022-07-25:
The client still assumes a running transaction because the last transaction hasn't been stopped properly. Now either the backend can stop it with a RemoteStopTransaction or you can stop it in the code by calling `stopTransaction()`. If you don't like that the charger preserves transactions over reboots, you can put `endSession()` into the setup function. It will ensure that the transaction will be stopped safely, if there is any.

The `id_tag_info` field is something which should originate in your server. It looks like the server creates a StartTransaction response without `idTagInfo` field which the JSON validator blocks. Maybe it helps to reset both the OCPP client and the server.

stig888881, 2022-07-25:
> The client still assumes a running transaction because the last transaction hasn't been stopped properly. Now either the backend can stop it with a RemoteStopTransaction or you can stop it in the [...]

If I understand correctly, the client does not send the idtaginfo field because it thinks there is a transaction going on? but I didn't call it earlier, how could it start? nothing came to the server.

matth-x, 2022-07-26:
The `idTag` and `idTagInfo` are two different things in OCPP. The client always sends the `idTag` and the server always responds with the `idTagInfo`.

The logs show that the client sent the `idTag` but received an error message:
```
[AO] To WS lib: [2,"1000040","StartTransaction",{"connectorId":1,"meterStart":0,"timestamp":"2022-07-25T15:49:24.000Z","idTag":" F6 CA 0C C5"}]
[AO] From WS lib: [4,"1000040","InternalError","An unexpected error occurred.",{}]
```

The server log shows that the server is actually missing the `idTagInfo`, but, it doesn't expect that field in the client request but looked for it in its own outbound response and couldn't find it. So it replied with an error message instead:

```
TypeError: init() missing 1 required positional argument: 'id_tag_info'
INFO:ocpp:ws://192.168.1.34: send [4,"1000040","InternalError","An unexpected error occurred.",{}]
```

Maybe the server is in an inconsistent state (e.g. it still has open transactions which you need to abort manually). I hope this helps to find the issue.

stig888881, 2022-07-26:
> The `idTag` and `idTagInfo` are two different things in OCPP. The client always sends the `idTag` and the server always responds with the `idTagInfo`.
>
> The logs show that the client sent the `idTag` [...]

Thanks a lot, such a simple mistake.

closed 2022-07-26 by stig888881