matth-x/MicroOcpp #73: How to add a meter_stop value

stig888881, 2022-07-26:
When I start a transaction: 
receive message [2,"1000239","StartTransaction",{"connectorId":1,"meterStart":0,"timestamp":"2022-07-26T21:34:10.000Z","idTag":" F6 CA 0C C5"}]
When I stop:
receive message [2,"1000242","StopTransaction",{"meterStop":0,"timestamp":"2022-07-26T21:34:12.000Z","transactionId":10001}]

The meter_stop field is 0, how can I change this value?

matth-x, 2022-08-02:
Hi,

The OCPP library sometimes stops transactions on its own. For example, if it receives the Reset command from the OCPP backend, it must stop all running transactions before rebooting. That's why the OCPP library needs a way to read the energy meter at any time and therefore, it needs a direct access to your energy meter.

You can give it the access by using this function:

https://github.com/matth-x/ArduinoOcpp/blob/48ddb71a7857cf2dbcf50c8fafcdde3694ad1792/src/ArduinoOcpp.h#L52

When setting up the microcontroller, you need to add a call like this

```
    setEnergyActiveImportSampler([]() {
        //read the energy register of the EVSE here and return the value in Watt-hours
        float energyRegister = ...
        return energyRegister;
    });
```

stig888881, 2022-08-04:
> Hi,
>
> The OCPP library sometimes stops transactions on its own. For example, if it receives the Reset command from the OCPP backend, it must stop all running transactions before rebooting. That's why [...]

Thks

closed 2022-08-04 by stig888881