matth-x/MicroOcpp #74: Project doesn't compile

arndt-s, 2022-07-28:
I am looking to bring ocpp on an Arduino board and found your project. Unfortunately I was not able to get it into a compiled version.

Checking out and building master fails with:
```
src/ArduinoOcpp/Core/ConfigurationContainerFlash.cpp:15:22: fatal error: LITTLEFS.h: No such file or directory
```

develop with:
```
collect2: error: ld returned 1 exit status
*** [.pio/build/nodemcuv2/firmware.elf] Error 1
```

Are there any specific framework versions or setup steps required that are not mentioned in the doc?
platformio version is up to date, environment clean.

Thank you.

matth-x, 2022-08-02:
Have you already been able to work around this problem?

Please try checking out from the develop branch. I have added a few things to improve the compatibility with different platforms.

Naheel-Azawy, 2022-08-07:
I'm having the same issue here. This seems to be a problem with esp8266 in particular. Compiling and uploading to esp32 seems to be working fine for me.
Dumb question, how to check out to develop in pio?
I tried fetching the full repo after pio installed it and then checkout to develop. Once pio runs, it notices a different hash and gets the one it's comfortable with.

Bhuvi-Naveen, 2023-05-21:
Hello,
I'm using ArduinoOcpp, with VS code and platformio, but when trying to compile, I'm getting linking error as below.
I'm new to VS code and platformio. can some please help

Processing esp32-development-board (platform: espressif32@3.5.0; board: esp-wrover-kit; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6) 
 - tool-esptoolpy @ 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.19.1
|-- WebSockets @ 2.3.6
|-- LittleFS_esp32 @ 1.0.5
|-- FS @ 1.0
|-- HTTPUpdate @ 1.3
Building in release mode
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\Configuration.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\ConfigurationContainer.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\ConfigurationContainerFlash.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\ConfigurationKeyValue.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\FilesystemAdapter.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\FilesystemUtils.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\OcppConnection.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\OcppEngine.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\OcppMessage.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\OcppModel.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\OcppOperation.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\OcppOperationTimeout.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\OcppServer.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\OcppSocket.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\OcppTime.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\OperationStore.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Core\OperationsQueue.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\Authorize.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\BootNotification.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\ChangeAvailability.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\ChangeConfiguration.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\ClearCache.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\ClearChargingProfile.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\DataTransfer.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\DiagnosticsStatusNotification.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\FirmwareStatusNotification.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\GetCompositeSchedule.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\GetConfiguration.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\GetDiagnostics.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\Heartbeat.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\MeterValues.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\RemoteStartTransaction.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\RemoteStopTransaction.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\Reset.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\SetChargingProfile.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\StartTransaction.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\StatusNotification.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\StopTransaction.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\TriggerMessage.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\UnlockConnector.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\MessagesV16\UpdateFirmware.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\Platform.cpp.o
Compiling .pio\build\esp32-development-board\src\ArduinoOcpp\SimpleOcppOperationFactory.cpp.o
[... 95 lines cut ...]
Compiling .pio\build\esp32-development-board\FrameworkArduino\esp32-hal-uart.c.o
[... cut ...]
============================================================== [FAILED] Took 47.72 seconds ==============================================================

Environment              Status    Duration
-----------------------  --------  ------------
esp32-development-board  FAILED    00:00:47.717
========================================================= 1 failed, 0 succeeded in 00:00:47.717 ========================================================= 

 *  The terminal process "C:\Users\navin\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

matth-x, 2023-08-25:
Closing this due to inactivity

closed 2023-08-25 by matth-x