matth-x/MicroOcpp #78: Setup in Arduino IDE

EZBOT1, 2022-08-09:
I am having trouble compiling  Library with Arduino IDE with es8266 and even with esp32. Can anyone give some insight how to just run the sample program.
Getting LITTLEFS.h error
Even after adding the library , getting vfs.h error .

clederson-Ms, 2022-08-16:
Estou também com problema ao copilar, não encontrei nenhuma postagem parecida, alguém poderia ajudar por favor??

usage: esptool write_flash [-h] [--erase-all]

                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]

                           [--flash_mode {keep,qio,qout,dio,dout}]

                           [--flash_size FLASH_SIZE]

                           [--spi-connection SPI_CONNECTION] [--no-progress]

                           [--verify] [--encrypt]

                           [--encrypt-files <address> <filename> [<address> <filename> ...]]

                           [--ignore-flash-encryption-efuse-setting]

                           [--compress | --no-compress]

                           <address> <filename> [<address> <filename> ...]

esptool write_flash: error: argument <address> <filename>: Address "{upload.erase_cmd}" must be a number

esptool write_flash: error: argument <address> <filename>: Address "{upload.erase_cmd}" must be a number

matth-x, 2023-08-25:
Closing this due to inactivity

closed 2023-08-25 by matth-x