matth-x/MicroOcpp #81: support for more than one connector

VjSargunam, 2022-08-20:
Hi matt,
            Currently your program supports single connector model. suppose if i want to increase more than one connector means what changes has to make in the existing code.

Thanks 
VJSargunam

piyush9999, 2023-02-02:
Hello @VjSargunam have you figure out how to set the code for two connectors

Fareed99, 2023-03-01:
Hi @VjSargunam and @piyush9999,
Have any of you guys figured out how to implement two connectors? If so, I would be interested to know how.
Much thanks!

aagamsheth, 2023-04-12:
Hi @piyush9999, @VjSargunam , @Fareed99, 
I have used this library to set up a single connector charger and now working on a dual connector, if you guys figure it out or hear from the developer @matth-x, I would be interested to know how to implement it.

piyush9999, 2023-04-13:
Hey @aagamsheth there's a new updated library available with multiple connectors integrated

aagamsheth, 2023-04-13:
@piyush9999 , thank you.

matth-x, 2023-08-25:
Closing this due to inactivity

closed 2023-08-25 by matth-x

chandan-ultraviolette, 2023-11-06:
Hi @piyush9999 @aagamsheth , can you confirm what all changes you did for the one more connector. I think only below changes should be done in the MicroOcpp.cpp ??

[image]

[image]