matth-x/MicroOcpp #85: esp-tls issue

vishutux, 2022-09-02:
Thanks for creating such a wonderful program, 

I am using Using examples/esp-tls and My CA certificate is fine and 

Using 
Signature Algorithm: sha256WithRSAEncryption
Public-Key: (2028 bit)

Getting below error , despite setting my CA certificate in 

const char ENDPOINT_CA_CERT[] PROGMEM 

Error coming on connection is below,

[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -9984
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:31): Disconnected
[E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed

matth-x, 2022-09-10:
Hi, thank you for the feedback!

The TLS library complains that the credentials are not right yet. Can you check if:

- The time on the ESP is set
- The subject name in the certificate is the same as the hostname in your URL?

matth-x, 2023-08-25:
Closing this due to inactivity

closed 2023-08-25 by matth-x