matth-x/MicroOcpp #90: ESP32+SIM800 GSM/GPRS

sevelm, 2022-09-13:
Hello,
Without taking a closer look at your great project, I'm just wondering whether I could do the whole thing with a SIM800 module without WiFi or Ethernet? Do you think this is feasible? Do you have experience and what should be considered?

THX

matth-x, 2022-09-14:
Hi,

The "core" OCPP library doesn't care about the underlying networking technology, as long as it gets an appropriate access to the networking stack.

To make getting started with this library easier, this GitHub project already comes with a network integration, namely [arduinoWebSockets](https://github.com/Links2004/arduinoWebSockets) and its underlying stack. However, that projects is not coupled with the OCPP client and you can exchange the WebSocket implementation as you wish.

That means that you have two options for integrating the SIM800:

- Create a new WebSockets driver based on a SIM800 connection by subclassing [`OcppSocket`](https://github.com/matth-x/ArduinoOcpp/blob/master/src/ArduinoOcpp/Core/OcppSocket.h)
- Continue using [arduinoWebSockets](https://github.com/Links2004/arduinoWebSockets), but route its traffic over the SIM800. Maybe it's worth checking out the [TinyGSM](https://github.com/vshymanskyy/TinyGSM) project. Both TinyGSM and arduinoWebSockets rely on the Arduino Client interface, so maybe the WebSockets project allows replacing the Wi-Fi client with the TinyGSM client.

piyush9999, 2022-10-03:
hey @matth-x @sevelm I think the first option is better to use by subclassing the OcppSocket, I have tried to route the arduinowebsocket traffic over GSM but the arduinoWebsocket library relies on the core esp32 wifi so it pushes DNS failed error.

While I am able to make websocket connection using both AT command and Tinygsm, so I need help futher how to use this code to subclass the OcppSocket. 
(the websocket connection is established on sim module after performing the get request and receiving websocket response headers from the server)

ankitdubeyskit, 2022-10-08:
Hi @piyush9999, I am also doing the same can you please help me in subclassing the OcppSocket. I have established the websocket connection to server successfully but I don't understand further what to do.
I am using tinyGSM library and modified the HTTP connection like as

    client.connect(host,port);
    client.print(String("GET ") + url + " HTTP/1.1\r\n");
    client.print(String("Host: ") + host+ "\r\n");
    client.print("Upgrade: websocket\r\n");
    client.print("Connection: Upgrade\r\n");
    client.print("Sec-WebSocket-Key: x3JJHMbDL1EzLkh9GBhXDw==\r\n");
    client.print("Sec-WebSocket-Protocol: ocpp2.0, ocpp1.6\r\n");
    client.print("Sec-WebSocket-Version: 13\r\n\r\n");
    client.println();     

I used this piece of code in websocket.begin(). I got the result "101 switching protocol". Please help what to do next. Is there any AT command based websocket library present? kindly share the link if available.
And many thanks to @matth-x for wonderful ocpp library.
Please guide @matth-x @piyush9999

piyush9999, 2022-10-08:
@ankitdubeyskit not yet, I'm curently using this library as reference and making a whole new code using Tinygsm

piyush9999, 2022-10-08:
Hello Ankit Sir, Piyush here, I'm also working on the Arduinoocpp and
trying to work it on gsm network.

Will it be okay if we get connect on WhatsApp

This is my whatsapp number:- 8605395447

Or you can share yours too

Thank You

piyush9999, 2022-10-08:
@ankitdubeyskit are you able to do it with AT command as well, because I have tried to add the AT sequence inside Arduino OCPP library, it was working but the corearduinowebsocket library is calling core wifi

ankitdubeyskit, 2022-10-09:
Hi Piyush,
My contact no is 7891485344. Sorry for late reply.

Fareed99, 2023-01-16:
I am working on the same too (connecting an ESP32 based client to a WebSockets server using GSM internet connectivity).
@ankitdubeyskit, did you manage to solve it somehow? Or maybe if @matth-x or @sevelm have any progress on this to share.
Would be highly obliged!

Thanks in advance!

matth-x, 2023-01-16:
Handling WebSocket connections over a GSM modem is a quite non-trivial problem. Sorry that I cannot provide a driver for it at the moment. But if you develop a driver which establishes and maintains the WebSocket, I can advise you how to link it to the OCPP library. And if you can imagine to publish the driver, I will promote it in the README of this project. I'm sure it could help a lot of people.

closed 2023-08-25 by matth-x