matth-x/MicroOcpp #93: How do i implement DataTransfer?

RiffuDev, 2022-09-29:
Hi,
I can see that you mentioned that dataTransfer and triggermessage feature is available. But it's just i can't able to find any Function to implement that..!! Please find the serialdebug below:

`
17:39:10.222 -> [AO] ERROR (lib\ArduinoOcpp\src\ArduinoOcpp\Core\OcppMessage.cpp:41): Unsupported operation: processReq() is not implemented
17:39:10.222 -> [AO] ERROR (lib\ArduinoOcpp\src\ArduinoOcpp\Core\OcppMessage.cpp:45): Unsupported operation: createConf() is not implemented
17:39:10.222 -> [AO] warning (lib\ArduinoOcpp\src\ArduinoOcpp\Core\OcppOperation.cpp:300): Operation failed. JSON CallError message: [4,"518ba0df-dc70-40cf-8ed8-9b91ca573147","NotImplemented","",{}]
`

BogdanFalk, 2022-10-21:
Any luck with this?

RiffuDev, 2022-10-21:
> Any luck with this?

Nope, i dont think theres any implementation for those here. But it says it is supported in the documentation. 
would appreciate if @matth-x replayed :)

matth-x, 2022-10-21:
Hi, sorry for the late response.

You can customize the DataTransfer by adding your own message class which implements the desired behavior.

#### 1. Define a new subclass of `OcppMessage`:

```cpp
#include <ArduinoOcpp/Core/OcppMessage.h>

class MyDataTransfer : public ArduinoOcpp::OcppMessage {
private:
    //...

public:
    MyDataTransfer(/* ... */);
    ~MyDataTransfer() = default; //or custom destructor

    const char* getOcppOperationType() override {return "DataTransfer";}
    
    //Alternative 1: implement the "initiate DataTransfer" functionality here
    std::unique_ptr<DynamicJsonDocument> createReq() override;
    void processConf(JsonObject payload) override;
    
    //Alternative 2: implement the "receive DataTransfer" functionality here
    void processReq(JsonObject payload) override; //get request
    virtual std::unique_ptr<DynamicJsonDocument> createConf() override; //send response
};
```

#### 2. If you want to initiate a DataTransfer message, implement the two "Alternative 1" functions above.

To send the custom DataTransfer message, wrap it into an `OcppOperation` object and pass it to the `OcppEngine` object:

```cpp
#include <ArduinoOcpp.h>
#include <ArduinoOcpp/Core/OcppEngine.h>
//...

    //trigger the DataTransfer
    auto op = ArduinoOcpp::makeOcppOperation(new MyDataTransfer(/* ... */));
    getOcppEngine()->initiateOperation(std::move(op));
```

Currently, `getOcppEngine()` is only available on the develop branch. If you use the master branch, please add a function to ArduinoOcpp.cpp for retrieving its `ocppEngine` object

#### 3. If you want to react on a DataTransfer message which comes from the server, implement the two "Alternative 2" functions above.

To let the OCPP library know about your customized message type, register it like that:

```cpp
#include <ArduinoOcpp/SimpleOcppOperationFactory.h>
//...

    //in setup()
    ArduinoOcpp::registerCustomOcppMessage("DataTransfer", [] () {
        return new MyDataTransfer(/* ... */);
    },
    nullptr);
```

#### In general

- OCPP specifies how the data must be structured inside the DataTransfer payload. With this custom approach, my library won't take over any of that. The OCPP specification explains the required data structure.
- For examples of how to create the `DynamicJsonDocument` containing the payload data or how to interpret the payload object that comes from the library, please refer to any predefined classes inside the MessagesV16 folder
- If you choose only one of the Alternatives above, remove the two declarations of the other alternatives as they won't be needed

TDVSVI, 2023-02-21:
> Hi, sorry for the late response.
> 
> You can customize the DataTransfer by adding your own message class which implements the desired behavior.
> 
> #### 1. Define a new subclass of `OcppMessage`:
> ```c++
> #include <ArduinoOcpp/Core/OcppMessage.h>
> 
> class MyDataTransfer : public ArduinoOcpp::OcppMessage {
> private:
>     //...
> 
> public:
>     MyDataTransfer(/* ... */);
>     ~MyDataTransfer() = default; //or custom destructor
> 
>     const char* getOcppOperationType() override {return "DataTransfer";}
>     
>     //Alternative 1: implement the "initiate DataTransfer" functionality here
>     std::unique_ptr<DynamicJsonDocument> createReq() override;
>     void processConf(JsonObject payload) override;
>     
>     //Alternative 2: implement the "receive DataTransfer" functionality here
>     void processReq(JsonObject payload) override; //get request
>     virtual std::unique_ptr<DynamicJsonDocument> createConf() override; //send response
> };
> ```
> 
> #### 2. If you want to initiate a DataTransfer message, implement the two "Alternative 1" functions above.
> To send the custom DataTransfer message, wrap it into an `OcppOperation` object and pass it to the `OcppEngine` object:
> 
> ```c++
> #include <ArduinoOcpp.h>
> #include <ArduinoOcpp/Core/OcppEngine.h>
> //...
> 
>     //trigger the DataTransfer
>     auto op = ArduinoOcpp::makeOcppOperation(new MyDataTransfer(/* ... */));
>     getOcppEngine()->initiateOperation(std::move(op));
> ```
> 
> Currently, `getOcppEngine()` is only available on the develop branch. If you use the master branch, please add a function to ArduinoOcpp.cpp for retrieving its `ocppEngine` object
> 
> #### 3. If you want to react on a DataTransfer message which comes from the server, implement the two "Alternative 2" functions above.
> To let the OCPP library know about your customized message type, register it like that:
> 
> ```c++
> #include <ArduinoOcpp/SimpleOcppOperationFactory.h>
> //...
> 
>     //in setup()
>     ArduinoOcpp::registerCustomOcppMessage("DataTransfer", [] () {
>         return new MyDataTransfer(/* ... */);
>     },
>     nullptr);
> ```
> 
> #### In general
> * OCPP specifies how the data must be structured inside the DataTransfer payload. With this custom approach, my library won't take over any of that. The OCPP specification explains the required data structure.
> * For examples of how to create the `DynamicJsonDocument` containing the payload data or how to interpret the payload object that comes from the library, please refer to any predefined classes inside the MessagesV16 folder
> * If you choose only one of the Alternatives above, remove the two declarations of the other alternatives as they won't be needed

Please tell me where I place these codes. I can't send and receive data from the DataTransfer function.

closed 2023-08-25 by matth-x

martamartz, 2024-01-29:
Hi,

How can I process a data transfer from the server in C?