matth-x/MicroOcpp #97: meter values updating every 4 minutes, how to bring back to 1 minute and how to add additional current & voltage measurends

nb4787, 2022-10-29:
OCPP is updating every 4 min and it should update every 1 minute

matth-x, 2022-10-31:
Hi, to add more measurands, refer to this example:

https://github.com/matth-x/ESP32_WiFi_V4.x/blob/cc5f5dff60ce423fec1370c26ecfdf56c85a22b5/src/ocpp.cpp#L151-L156

Note that you also need to activate the reading in OCPP by changing the configuration. The `MeterValuesSampledData` needs to contain the new measurand (e.g. `Current.Import`).

And yes, currently the library collects 4 measurements before sending them to the OCPP server. I will change the default behavior so it sends every measurement immediately. In the meantime you can lower the sample interval to 15s, then you will get the updated values every minute.

Feel free to reopen if this doesn't solve this issue.

closed 2022-10-31 by matth-x

nb4787, 2022-11-03:
Can you suggest how to change 4 minute to 1 minute for , for meter values update every minute to steve ?