matth-x/MicroOcpp #98: .json package not found while trying to compile ESP example for ESP8266

ritu20chauhan, 2022-10-31:
The code from the example of ESP was successfully compiled for ESP32.
But when I tried the same code for ESP8266 the was error "Package.Json" not found.
 
[image]

nb4787, 2022-10-31:
can you connect me mail id ? im also working with ocpp integration with esp32

nb4787, 2022-10-31:
cdacndb@gmail.com

matth-x, 2022-11-02:
Hi Ritu,

The package.json is not a part of this library, but belongs to the PlatformIO ecosystem. So probably your installation of the IDE is corrupted. In [this page](https://community.platformio.org/t/upload-error-could-not-find-one-of-package-json-manifest-files-in-the-package/28149) I found that reinstalling the toolchain could solve the issue.

closed 2022-11-02 by matth-x