matth-x/MicroOcpp #99: Can you suggest how to change 4 minute to 1 minute for , for meter values update every minute to steve ?

nb4787, 2022-11-03:
i need to steve should update metervalues every minute ? Can you suggest explain

matth-x, 2022-11-03:
There are two alternatives:

1. Change the Configuration `MeterValueSampleInterval` to 15s in your Central System. By default, the charger will send 4 MeterValues every minute then.
2. Alternatively, find [this line](https://github.com/matth-x/ArduinoOcpp/blob/10795aad7e124917a0336b4d98fd51b47c197453/src/ArduinoOcpp/Tasks/Metering/ConnectorMeterValuesRecorder.cpp#L23) in the source code. Change the 4 to 1, erase the flash memory and upload the firmware again.

closed 2022-11-03 by matth-x

matth-x, 2022-11-13:
03177f93e7d2c6b6b15c07b5d0f9944da65512b0 addresses this issue. You can change the custom configuration `AO_MeterValueCacheSize` to 1 and then the charger will send each measurement immediately to the server.

bananaslug123, 2022-11-13:
Anna, fyi I had to make a stop and traffic is tight.  I should be there by 4:30…Thanks,Brian