matth-x/MicroOcpp #102: Include optional fields.

fanzypantz, 2022-11-17:
Hi, is it possible to provide commands with optional fields. Such as timestamps on most things, transactionData on stopTransaction and so on? Or is this fairly rigid? 

As you might know, CPO's might be a bit picky on what you require with their systems.. For example monta doesn't even send an OK signal back if you don't include the transactionData, which sounds dumb as its "optional" defeating the purpose of the first "O" in OCPP...

matth-x, 2022-11-18:
Hi, normally this library omits optional fields if the underlying information does not exist. The OCPP documentation is very specific about it so unfortunately, there is no easy way to change that behavior.

I can think of two approaches. One is to fork this library and to add the desired fields manually in `createReq(...)` and `createConf(...)` in each class in the MessagesV16 folder. Hopefully it will still be easy to replay all updates of this library on your fork as most updates won't affect the message creation in future. The other approach would be to scan all output of this library, filter for the OCPP operations with optional fields and add default values for all optional fields which are not present yet. This can be done by decorating an `OcppSocket` with a filter (look for "decorator pattern" on Google). The latter approach would definitely work for all future updates of this library without too much maintenance.

matth-x, 2023-01-07:
Hi @fanzypantz, are there any updates about this?

matth-x, 2023-08-25:
Closing this due to inactivity

closed 2023-08-25 by matth-x