matth-x/MicroOcpp #103: What does return 0.f means

ritu20chauhan, 2022-11-24:
setEnergyMeterInput([]() {
        //take the energy register of the main electricity meter and return the value in watt-hours
        return 0.f;
[image]

matth-x, 2022-11-25:
`0.f` is a placeholder which you can replace with a meaningful value.

The EnergyMeterInput should read the value of the energy meter and return it to the OCPP library. For example, given you store the energy register value in the global variable `float e_reg`, then change the statement to `return e_reg;`. The idea is that the OCPP library can fetch the energy value any time by executing the Input function.

matth-x, 2023-01-07:
Closed due to inactivity. Please reopen for further questions.

closed 2023-01-07 by matth-x