matth-x/MicroOcpp #104: Stack overflow while running the esp-idf example

ysaakpr, 2022-12-02:
Trying to run the esp-idf example, as per today's latest idf version, and Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
Consistently getting reboot due to the stack overflow at the same place after the first connect response.
```
[AO] info (./components/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: /ocpp-creds.jsn
6bc 3 mongoose.c:3440:mg_connect        1 -1 ws://192.168.1.4:9000/test1
6c4 3 mongoose.c:4324:mg_connect_resolv 1 0x36 -> c0a80104:9000 pend
[AO] info (./components/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: /opstore.cnf
[AO] info (./components/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: /txstore.jsn
[AO] info (./components/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: /volatile
[AO] warning (./components/AOcppMongoose/src/AOcppMongooseClient.cpp:459): Insecure connection (WS)
805 3 mongoose.c:4252:write_conn        1 0x36 snd 188/2048 rcv 0/0 n=188 err=2
815 3 mongoose.c:4241:read_conn         1 0x36 snd 0/2048 rcv 0/2048 n=236 err=2
[AO] info (./components/AOcppMongoose/src/AOcppMongooseClient.cpp:463): connection ws://192.168.1.4:9000/test1 -- connected!
[AO] Send: [2,"1000000","BootNotification",{"chargePointModel":"ESP-IDF charger","chargePointVendor":"Your brand name here"}]
bd6 3 mongoose.c:4252:write_conn        1 0x36 snd 120/2048 rcv 0/2048 n=120 err=2
bdf 3 mongoose.c:4241:read_conn         1 0x36 snd 0/2048 rcv 0/2048 n=94 err=2
[AO] Recv: [3,"1000000",{"currentTime":"2022-12-02T15:36:30.851100","interval":10,"status":"Accepted"}]

***ERROR*** A stack overflow in task main has been detected.

Backtrace: 0x40081b52:0x3ffb9520 0x40088671:0x3ffb9540 0x4008bc6e:0x3ffb9560 0x4008a302:0x3ffb95e0 0x4008876c:0x3ffb9600 0x4008871e:0x3ffb9610 0x4008902b:0x40084e94 |<-CORRUPTED
0x40081b52: panic_abort at /Users/vysh/esp/esp-idf/components/esp_system/panic.c:402

0x40088671: esp_system_abort at /Users/vysh/esp/esp-idf/components/esp_system/esp_system.c:128

0x4008bc6e: vApplicationStackOverflowHook at /Users/vysh/esp/esp-idf/components/freertos/port/xtensa/port.c:394

0x4008a302: vTaskSwitchContext at /Users/vysh/esp/esp-idf/components/freertos/tasks.c:3504

0x4008876c: _frxt_dispatch at /Users/vysh/esp/esp-idf/components/freertos/port/xtensa/portasm.S:436

0x4008871e: _frxt_int_exit at /Users/vysh/esp/esp-idf/components/freertos/port/xtensa/portasm.S:231

0x4008902b: xQueueGenericSend at /Users/vysh/esp/esp-idf/components/freertos/queue.c:946

0x40084e94: esp_crosscore_int_send_yield at /Users/vysh/esp/esp-idf/components/esp_system/crosscore_int.c:145

ELF file SHA256: 90d6339b3dd34cfb

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6660
load:0x40078000,len:14848
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (29) boot: ESP-IDF v4.4.3 2nd stage bootloader
I (29) boot: compile time 15:17:22
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00010000 00190000
I (99) boot:  4 ota_1            OTA app          00 11 001a0000 00190000
I (106) boot:  5 ao               Unknown data     01 82 00330000 000d0000
I (114) boot: End of partition table
I (118) boot_comm: chip revision: 1, min. application chip revision: 0
I (125) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1e460h (124000) map
I (179) esp_image: segment 1: paddr=0002e488 vaddr=3ffb0000 size=01b90h (  7056) load
I (182) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=f47f8h (1001464) map
I (547) esp_image: segment 3: paddr=00124820 vaddr=3ffb1b90 size=01ccch (  7372) load
I (550) esp_image: segment 4: paddr=001264f4 vaddr=40080000 size=14978h ( 84344) load
I (587) esp_image: segment 5: paddr=0013ae74 vaddr=50000000 size=00010h (    16) load
I (598) boot: Loaded app from partition at offset 0x10000
I (598) boot: Disabling RNG early entropy source...
I (610) cpu_start: Pro cpu up.
I (610) cpu_start: Starting app cpu, entry point is 0x40081290
0x40081290: call_start_cpu1 at /Users/vysh/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (596) cpu_start: App cpu up.
I (626) cpu_start: Pro cpu start user code
I (626) cpu_start: cpu freq: 160000000
I (626) cpu_start: Application information:
I (631) cpu_start: Project name:     ao_example
I (636) cpu_start: App version:      1
I (640) cpu_start: Compile time:     Dec  2 2022 21:05:26
I (647) cpu_start: ELF file SHA256:  90d6339b3dd34cfb...
I (652) cpu_start: ESP-IDF:          v4.4.3
I (658) heap_init: Initializing. RAM available for dynamic allocation:
I (665) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (671) heap_init: At 3FFB7A38 len 000285C8 (161 KiB): DRAM
I (677) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (683) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (690) heap_init: At 40094978 len 0000B688 (45 KiB): IRAM
I (697) spi_flash: detected chip: generic
I (701) spi_flash: flash io: dio
I (706) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (771) wifi station: ESP_WIFI_MODE_STA
I (781) wifi:wifi driver task: 3ffc0760, prio:23, stack:6656, core=0
I (781) system_api: Base MAC address is not set
I (781) system_api: read default base MAC address from EFUSE
I (801) wifi:wifi firmware version: 8cb87ff
I (801) wifi:wifi certification version: v7.0
I (801) wifi:config NVS flash: enabled
I (801) wifi:config nano formating: disabled
I (801) wifi:Init data frame dynamic rx buffer num: 32
I (811) wifi:Init management frame dynamic rx buffer num: 32
I (811) wifi:Init management short buffer num: 32
I (821) wifi:Init dynamic tx buffer num: 32
I (821) wifi:Init static rx buffer size: 1600
I (821) wifi:Init static rx buffer num: 10
I (831) wifi:Init dynamic rx buffer num: 32
I (831) wifi_init: rx ba win: 6
I (841) wifi_init: tcpip mbox: 32
I (841) wifi_init: udp mbox: 6
I (841) wifi_init: tcp mbox: 6
I (851) wifi_init: tcp tx win: 5744
I (851) wifi_init: tcp rx win: 5744
I (861) wifi_init: tcp mss: 1440
I (861) wifi_init: WiFi IRAM OP enabled
I (861) wifi_init: WiFi RX IRAM OP enabled
I (871) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (971) wifi:mode : sta (98:f4:ab:62:0a:94)
I (971) wifi:enable tsf
I (971) wifi station: wifi_init_sta finished.
I (981) wifi:new:<8,0>, old:<1,0>, ap:<255,255>, sta:<8,0>, prof:1
I (991) wifi:state: init -> auth (b0)
I (991) wifi:state: auth -> assoc (0)
I (1001) wifi:state: assoc -> run (10)
I (1021) wifi:connected with Danger, aid = 21, channel 8, BW20, bssid = f0:2f:74:7a:40:a1
I (1021) wifi:security: WPA2-PSK, phy: bgn, rssi: -28
I (1031) wifi:pm start, type: 1

I (1051) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (1261) wifi:<ba-add>idx:0 (ifx:0, f0:2f:74:7a:40:a1), tid:0, ssn:1, winSize:64
```

matth-x, 2022-12-02:
Hi, thanks for letting me know. The default stack size isn't large enough. You can find here how to increase it: [https://github.com/espressif/esp-idf/issues/2824#issuecomment-566865807](https://github.com/espressif/esp-idf/issues/2824#issuecomment-566865807). 997d6b4405c6daa87236e2850945de97c0140015 addresses this.

Did you have any other difficulties with setting up the ESP-IDF example? The guide is relatively new and I'm happy if I can improve it.

ysaakpr, 2022-12-02:
I am trying it today.. will give you an update if I am getting any trouble.. thanks for the quick update. You have done amazing work through this project. Happy to contribute.. Though I am immature in embedded programming

ysaakpr, 2022-12-02:
The above-mentioned fix worked.

closed 2022-12-02 by ysaakpr