matth-x/MicroOcpp #105: problem while compiling the code

ritu20chauhan, 2022-12-10:
while compiling the code there are several problems like "sys/cdefs.h" file can't be opened, "serial" is not defined, and so on. 
Can anyone help me in understanding why is this occurring?
[image]
[image]

nb4787, 2022-12-12:
connect for development of ocpp 1.6j nb4787@gmail.com

matth-x, 2022-12-12:
Hm, the build system seems to be messed up somehow ... neither the include statements nor the build flags work.

The only suspicious think I can see in your screenshots is the `.vscode` folder in the root directory. In my VSCode workspace it is only inside project folders but never on the same hierarchy level like the projects. Maybe it is worth trying to delete the `.vscode` folder or starting with a fresh VSCode workspace?

matth-x, 2023-02-02:
Closing this due to inactivity. If you found a solution and want to help others with a similar issue, feel free to post it here.

closed 2023-02-02 by matth-x