matth-x/MicroOcpp #106: Not getting compiling and uploading icon in platform.io

ritu20chauhan, 2022-12-10:
while using platform.io for arduinoOcpp, the platform does not shows compiling & uploading symbol.
These are visible when I am running some other code like- connecting esp to wifi basics.
This problem while using arduinoOcpp in platform.io happens frequently, so can anybody let me know why does this happen? Is there any installation
[image]
 problem or some changes are required in system?

matth-x, 2022-12-12:
What happens if you open the project ArduinoOcpp-master directly via "Open Folder..."?

matth-x, 2023-01-07:
Closed due to inactivity. Please open a new issue for follow-up questions.

closed 2023-01-07 by matth-x