matth-x/MicroOcpp #107: creating a network layer in socket.cpp

nb4787, 2022-12-24:
hi matth-x thank you for creating ultimate library for ocpp 1.6j 
can you please explain how to create network layer with gsm simcom module , when i was connecting through gsm network 
its trying to connect websocket with wifi connection 
please ex[lain how divert root connection gsm module @matth-x

matth-x, 2023-01-07:
The included WebSockets integration is optimized towards Wi-Fi connections. For GSM I would recommend developing a new driver which creates and maintains a WebSocket connection over GSM. As soon as that connection is stable, the next step is to create a separate adapter class which talks to your WebSocket driver and inherits from OcppSocket so you will be able to pass it to the OCPP library during intialization.

I would highly recommend to start the development without the OCPP lib but with the WebSockets library only (btw. you can choose whatever WS lib you like). But please understand that this is out of scope of this open source project and I cannot provide support for it.

Btw. I'm seeking contributions! If you want to contribute back the GSM driver, please let me know.

closed 2023-01-07 by matth-x

Fareed99, 2023-01-16:
@nb4787 did you manage to solve this? If so, I would be very interested to know how as I am currently facing the same issue.

Thanks in advance!

nb4787, 2023-01-16:
Connect with me nb4787@gmail.com

Fareed99, 2023-01-16:
@nb4787 I did. Waiting for your response.
Thanks!

matth-x, 2023-01-16:
See https://github.com/matth-x/ArduinoOcpp/issues/90#issuecomment-1384224412