matth-x/MicroOcpp #110: addMeterValueInput only works for power and energy

pedro-fuoco, 2023-01-05:
Hi @matth-x ,
I've been trying to use the new addMeterValueInput function that was added on v0.2.0, but I'm having some issues...

The MeterValues added with the measurand "Power.Active.Import" or "Energy.Active.Import.Register" work perfectly fine, but every other possible measurand isn't sent on the MeterValue.req to the Central System. Here's the code:

```
addMeterValueInput([](){return readElectricPower();}, "Power.Active.Import", "kW", nullptr, nullptr, 1);
addMeterValueInput([](){return readElectricEnergy();}, "Energy.Active.Import.Register", "kWh", nullptr, nullptr, 1);

addMeterValueInput([](){return readElectricTension(0);}, "Voltage", "V", nullptr, "L1", 1);
addMeterValueInput([](){return readElectricTension(1);}, "Voltage", "V", nullptr, "L2", 1);
addMeterValueInput([](){return readElectricTension(2);}, "Voltage", "V", nullptr, "L3", 1);

addMeterValueInput([](){return readElectricCurrent(0);}, "Current.Import", "A", nullptr, "L1", 1);
addMeterValueInput([](){return readElectricCurrent(1);}, "Current.Import", "A", nullptr, "L2", 1);
addMeterValueInput([](){return readElectricCurrent(2);}, "Current.Import", "A", nullptr, "L3", 1);
```

Thank you for the attention, I am only asking here because I've tried multiple things and looked through the whole documentation and still can't find the answers.

matth-x, 2023-01-07:
Hi @pedro-fuoco,

That code looks good, now the OCPP server should be able to query those readings. Can you send a `ChangeConfiguration` request to the charger for the key `MeterValuesSampledData` and add the new measurands to the value? The new value `Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import` should do the trick.

pedro-fuoco, 2023-01-09:
Thank you so much for the quick reply!
I'm investigating the MeterValuesSampledData configuration now.
Would there be a way to change the MeterValuesSampledData key "default value" from within the charger ? Multiple chargers will use this code and we don't want to need to send a ChangeConfiguration request for every new one whenever they are created. 
I imagine this is possible since ABB chargers, for example, don't need to receive this request from the Central System.
Thanks again!

matth-x, 2023-01-10:
Yes, it's possible and it's a reasonable thing to change the default values.

Each OCPP configuration is stored in a separate data object (AbstractConfiguration as you name it and its concrete subclasses, i.e. Configuration<const char*> and so on). To fetch that data object and update it, use this code:

```cpp
std::shared_ptr<ArduinoOcpp::Configuration<const char *>> mvSampledData = ArduinoOcpp::declareConfiguration<const char *>(
        "MeterValuesSampledData", "", CONFIGURATION_FN);
const char *example = *mvSampledData; //example: read value of configuration
*mvSampledData = "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import"; //example: update configuration value
ArduinoOcpp::configuration_save(); //write configs file back to flash immediately (not really necessary though)
```

Setting factory defaults is a little trickier: once `declareConfiguration` is called the first time for a certain configuration, it applies the factory defaults. So you need to declare the configuration before `OCPP_initialize` which would set the built-in defaults. To set custom factory defaults, use this code:

```cpp
#include <ArduinoOcpp/Core/Configuration.h>
```
```cpp
std::shared_ptr<ArduinoOcpp::FilesystemAdapter> filesystem = ArduinoOcpp::makeDefaultFilesystemAdapter(ArduinoOcpp::FilesystemOpt::Use_Mount_FormatOnFail);
ArduinoOcpp::configuration_init(filesystem);

ArduinoOcpp::declareConfiguration<const char *>(
        "MeterValuesSampledData", "DEFAULT VALUE", CONFIGURATION_FN); //declare configuration and use the default value parameter (replace with required value)
ArduinoOcpp::configuration_save(); //again, not really necessary

OCPP_initialize(...); //initialize library normally
```

The code above can be executed during each start up, but it only has an effect if the flash is still empty, e.g. during the first run. To test it, erase the flash before.

pedro-fuoco, 2023-01-10:
This absolutelly did the trick!

Thank you for the insight and examples

closed 2023-01-10 by pedro-fuoco

mukeshtalks, 2023-02-02:
@pedro-fuoco I am also wondering how and where to add the New parameter in meter value inputs like SOC, Voltage, Current etc. Can you guide me on how and where to add these as you have already done? Can you specify the file and function that I need to modify?. Your support will really help me to proceed further.

matth-x, 2023-02-02:
See this: https://github.com/matth-x/ArduinoOcpp/issues/114

pedro-fuoco, 2023-03-16:
@mukeshtalks I just added the following code right after using the addMeterValueInput
```
std::shared_ptr<ArduinoOcpp::Configuration<const char *>> mvSampledData = ArduinoOcpp::declareConfiguration<const char *>(
        "MeterValuesSampledData", "", CONFIGURATION_FN);
*mvSampledData = "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import";
ArduinoOcpp::configuration_save();
```