matth-x/MicroOcpp #112: SampleValues as floats

pedro-fuoco, 2023-01-26:
Hi @matth-x, 
I've been using the addMeterValueInput function extensively in the last month, thank you for adding it and helping me figure it out in the past.

Recently I've been wondering why does it only accept int32_t functions as input ``` void addMeterValueInput(std::function<int32_t ()> valueInput, const char *measurand, const char *unit, const char *location, const char *phase, unsigned int connectorId) ``` instead of floats like ```void setPowerMeterInput(std::function<float()> powerInput, unsigned int connectorId)``` for example.

Transforming float reading values of voltage and current into int32 has impacted my application significantly, is there a way to work around this ? I would like to send the whole float value to the Central System and I figure this should be possible since the ocpp1.6 protocol reads as following when describing SampledValues:
"Required. Value as a “Raw” (decimal) number or “SignedData”. Field Type is “string” to allow for digitally signed data readings. Decimal numeric values are also acceptable to allow fractional values for measurands such as Temperature and Current"

pedro-fuoco, 2023-01-27:
I have noticed that the message that reaches the Central System presents exclusivelly integers, even though my metering functions return floats
```
{"connectorId":1,"transactionId":95216,"meterValue":[{"timestamp":"2023-01-27T18:52:31.000Z","sampledValue":
[{"value":"11","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"},
{"value":"0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},
{"value":"0","context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V"},
{"value":"0","context":"Sample.Periodic","measurand":"Voltage","phase":"L2","unit":"V"},
{"value":"131","context":"Sample.Periodic","measurand":"Voltage","phase":"L3","unit":"V"},
{"value":"0","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A"},
{"value":"0","context":"Sample.Periodic","measurand":"Current.Import","phase":"L2","unit":"A"},
{"value":"0","context":"Sample.Periodic","measurand":"Current.Import","phase":"L3","unit":"A"}]}]}
```

matth-x, 2023-02-02:
Hi @pedro-fuoco, sorry for the late reply.

You're right here, the facade lacks floating point inputs. The reason is that integer numbers are far more important and the API shouldn't get too bloated. If I get more requests like that I will add a built-in function. You can add a custom data type for this though:

1. Create a serializer and deserializer for the custom data type (here: `float`):

```cpp
class FloatDeSerializer {
public:
    static float deserialize(const char *str) {
        //TODO convert string to float
        return 0.0f;
    }
    static bool ready(float val) {
        return true;
    }
    static std::string serialize(float val) {
        //TODO convert float to string
        return "";
    }
    static int32_t toInteger(float val) {
        return (int32_t) val;
    }
};
```

2. Register the MeterValueInput:

```cpp
setup() {
    //...

    ArduinoOcpp::SampledValueProperties properties;
    properties.setMeasurand("Current.Import");
    properties.setLocation("Outlet"); //optional
    properties.setUnit("A"); //optional
    properties.setPhase("L1"); //optional
    auto reader = new ArduinoOcpp::SampledValueSamplerConcrete<int32_t, FloatDeSerializer>(properties, [] (ArduinoOcpp::ReadingContext) -> float {return current_reg;});
    addMeterValueInput(std::unique_ptr<ArduinoOcpp::SampledValueSamplerConcrete<int32_t, FloatDeSerializer>>(reader));

    // ...
}
```

And then it should be done.

pedro-fuoco, 2023-02-08:
It works wonders! Thank you so much @matth-x.
For anyone interested this is how I serialized/deserialized the floats:

```
class FloatDeSerializer {
    public:
        static float deserialize(const char *str) {
            return atof(str);
        }
        static bool ready(float val) {
            return true;
        }
        static std::string serialize(float val) {
            char buffer[20];
            dtostrf(val,4,9,buffer);
            return buffer;
        }
        static int32_t toInteger(float val) {
            return (int32_t) val;
        }
};
```
I get the argument of not bloating the API, but many Central Systems require precise measurements to interpret them later. In applications like this I really recommend not rounding the numbers down to an integer as it can become a massive problem in analysis tools.

pedro-fuoco, 2023-02-08:
@matth-x if you would like I can integrate this into the source code and submit a pull request, what do you think?

closed 2023-02-08 by pedro-fuoco

matth-x, 2023-02-08:
@pedro-fuoco That would actually be pretty cool!

The built-in solution needs one modification, it should be the `<float>` template specialization of the `SampledValueDeSerializer` and follow after this:

https://github.com/matth-x/ArduinoOcpp/blob/27ce8b15cd5158e1b8d637b67fe55416093bfe57/src/ArduinoOcpp/Tasks/Metering/SampledValue.h#L23-L34

And I think after adding the API function in ArduinoOcpp.h and .cpp, it could be done. Maybe it needs some more changes, but I'm sure you will find the missing parts.