matth-x/MicroOcpp #113: How to simulate end to end transaction with remote transaction without connecting with vehicle

mukeshtalks, 2023-01-27:
Hello all, I have compiled the example examples/ESP/main.cpp and am able to connect with the OCPP server. I have changed in the call-back setConnectorPluggedInput(cb) and sent changed the status from false to true with the help of GPIO. find the code below for your reference.

setConnectorPluggedInput([]() {
        //return true if an EV is plugged to this EVSE
        if(!digitalRead(CONNECTOR))
        {
            digitalWrite(LED, HIGH);
            return true; 
        }
        else
        {
            digitalWrite(LED, LOW);
            return false; 
        }
    });

when I am returning True it sends the preparing status to the OCPP server. When I send the Remote start transaction from the OCPP server, it is not entering into the start charging state. Check the logs below.

[AO] Recv: WS pong
[AO] info (src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:50): New status: Available (connectorId 1)
[AO] Send: [2,"1000007","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2023-01-27T11:03:56.000Z"}]
[AO] Recv: [3,"1000007",{}]
[AO] Recv: WS ping
[AO] Recv: WS pong
[AO] Recv: [2,"d78f65b4-eabe-4ab9-9a1b-ec964dc29254","RemoteStartTransaction",{"connectorId":1,"idTag":"04B0267AE05C10"}]
[AO] Send: [3,"d78f65b4-eabe-4ab9-9a1b-ec964dc29254",{"status":"Accepted"}]
[AO] info (src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:50): New status: Preparing (connectorId 1)
[AO] Send: [2,"1000008","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2023-01-27T11:04:18.000Z"}]
[AO] Recv: [3,"1000008",{}]
[AO] Recv: WS pong

Can anyone help me with How I can simulate the remote transaction with the help of simulated values?

piyush9999, 2023-02-02:
Hello @mukeshtalks could you please help me in adding multiconnector functionality

matth-x, 2023-02-02:
Hi, if you don't have a connector plug, you can also remove the `setConnectorPluggedInput` from your source code completely. Then it will start the transaction immediately after receiving the RemoteStartTransaction, without the need to simulate a plug.

closed 2023-02-02 by matth-x