matth-x/MicroOcpp #114: How to Add the new parameters in the Meter values like SOC, Voltage, Current and Temeprature

mukeshtalks, 2023-02-02:
Hello All, I am able to work with my CMS and geting the Energy (Wh) and the Power while charging. I need to Add More parameters in the meter reading like SOC, Voltage, Current and temperature, etc. 
Need the know how I can add these to the code. If any document or example is available or any guidance on how I can add that will be great.

mukeshtalks, 2023-02-02:
@piyush9999 I have not tried yet. I was working with one connector only. Can you help me if you are able to add the parameters I am looking for.

matth-x, 2023-02-02:
Hi, you can use the more generic function `addMeterValueInput`. It works similar to the EneryInput and PowerInput, but you can define more details. See the example here:

https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/751d8374ab072df83a920295b6723dc8e904d21e/src/ocpp.cpp#L165-L169

closed 2023-02-02 by matth-x

matth-x, 2023-02-02:
@piyush9999 Sorry, I forgot to mention multiple connectors here. See the following code snippet:

```cpp
int connectorId = 1; //select connectorId 0, 1 or 2

addMeterValueInput([this] () {
    return (int32_t) evse->getAmps();
}, 
"Current.Import",
"A",
nullptr, //location (optional)
nullptr, //phase (optional)
connectorId);
```

piyush9999, 2023-02-03:
Hey, @matth-x if I want to use Two connectors, then the statusNotification should send connector id 1 and 2 but in current condition, it is sending 0 and 1. Is there anything in the main source code that I have to add to get multiple connectors functionality. So I can start and stop transaction and meter them separately

matth-x, 2023-02-03:
Hi @piyush9999, in platformio.ini, add `-D AO_NUMCONNECTORS=3` to the `build_flags` section.

The first connector has connectorId 1, the second has Id 2. You can ignore Id 0 for the moment.

mukeshtalks, 2023-02-06:
@matth-x thank you for your reply. I will try the solution you suggest and get back to you if any issues I will face.

piyush9999, 2023-02-06:
hey @matth-x I have successfully added two connectors 1 and 2 as you told.

So when I'm starting transaction for Connector ID 2 I'm getting an error from the library as MeterStart Undefined

Log:
[AO] Recv: [2,"a992d12c-cfbd-429d-94df-25382d5a52ff","RemoteStartTransaction",{"connectorId":2,"idTag":"Abcd12345"}]
17:34:56.138 -> [AO] Send: [3,"a992d12c-cfbd-429d-94df-25382d5a52ff",{"status":"Accepted"}]
17:34:56.138 -> [AO] info (lib_extras/ArduinoOcpp-master/src/ArduinoOcpp/Tasks/ChargePointStatus/ConnectorStatus.cpp:237): Session mngt: trigger StartTransaction
17:34:56.138 -> [AO] ERROR (lib_extras/ArduinoOcpp-master/src/ArduinoOcpp/Tasks/ChargePointStatus/ConnectorStatus.cpp:245): MeterStart undefined

And same error when I'm stopping the transaction
Log:
[AO] ERROR (lib_extras/ArduinoOcpp-master/src/ArduinoOcpp/Tasks/ChargePointStatus/ConnectorStatus.cpp:301): MeterStop undefined

matth-x, 2023-02-06:
@piyush9999 When adding more connectors, it is necessary to add the setup functions for each connector again. For example, when setting the energy meter:

https://github.com/matth-x/ArduinoOcpp/blob/27ce8b15cd5158e1b8d637b67fe55416093bfe57/src/ArduinoOcpp.h#L191

you need to execute that function once for connectorId = 1 and for connectorId = 2. The same is required for each function which has the parameter `unsigned int connectorId`.

mukeshtalks, 2023-02-06:
@matth-x I have added the meter values as suggested by you but these values are not getting added to the meter values while charging. Can you please suggest what could be wrong? I have added the meter values in main.cpp file in Setup function in SECC example in the library.

mukeshtalks, 2023-02-06:
@matth-x I could able to send the meter values to my CMS with new inputs. I missed adding these new inputs to default configurations. After adding these this is working. Thank you for your Support.

piyush9999, 2023-02-06:
@matth-x Thanks brother added☺

mukeshtalks, 2023-02-17:
@matth-x I have added the code as given below in the setup to add the new measurement values. But after adding this code I started facing an issue. At every power up I need to configure the WebSocket again. Please suggest where could be the issue, or I am missing something. 

std::shared_ptr<ArduinoOcpp::Configuration<const char *>> mvSampledData = ArduinoOcpp::declareConfiguration<const char *>(
        "MeterValuesSampledData", "", CONFIGURATION_FN);
    const char *example = *mvSampledData; //example: read value of configuration
    *mvSampledData = "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import,SoC,Temperature"; //example: update configuration value
    //ArduinoOcpp::configuration_save(); //write configs file back to flash immediately (not really necessary though)

 std::shared_ptr<ArduinoOcpp::FilesystemAdapter> filesystem = ArduinoOcpp::makeDefaultFilesystemAdapter(ArduinoOcpp::FilesystemOpt::Use_Mount_FormatOnFail);
    ArduinoOcpp::configuration_init(filesystem);
#if 1
    ArduinoOcpp::declareConfiguration<const char *>(
        "MeterValuesSampledData", "DEFAULT VALUE", CONFIGURATION_FN); //declare configuration and use the default value parameter (replace with required value)
    //ArduinoOcpp::configuration_save(); //again, not really necessary
#endif

mukeshtalks, 2023-02-17:
@matth-x below is the code in main.cpp for your reference. 

`// matth-x/ArduinoOcpp
// Copyright Matthias Akstaller 2019 - 2022
// MIT License
//
//
// IMPORTANT: Please add https://github.com/tzapu/WiFiManager.git to the lib_deps of your project!
//
// This sketch demonstrates a complete OCPP Wi-Fi module of an EVSE. You can flash
// an ESP8266/ESP32 with this program, build it into your EVSE and start charging with
// OCPP-connectivity.
//
// The pin mapping and additional comments can be found in this file. You probably need
// to adapt a few things before deployment.
//

#include <Arduino.h>
#include <WiFiManager.h> //please add to lib_deps: https://github.com/tzapu/WiFiManager.git

#include <ArduinoOcpp.h>
#include <ArduinoOcpp/Core/Configuration.h> //load and save settings of WiFi captive portal

/*
 * Pin mapping part I
 * Interface to the SECC (Supply Equipment Communication Controller, e.g. the SAE J1772 module)
 */
#define AMPERAGE_PIN 4 //modulated as PWM

#define EV_PLUG_PIN 14 // Input pin | Read if an EV is connected to the EVSE
#if defined(ESP32)
#define EV_PLUGGED HIGH
#define EV_UNPLUGGED LOW
#elif defined(ESP8266)
#define EV_PLUGGED LOW
#define EV_UNPLUGGED HIGH
#endif

#define OCPP_CHARGE_PERMISSION_PIN 5 // Output pin | Signal if OCPP allows / forbids energy flow
#define OCPP_CHARGE_PERMITTED HIGH
#define OCPP_CHARGE_FORBIDDEN LOW

#define EV_CHARGE_PIN 12 // Input pin | Read if EV requests energy (corresponds to SAE J1772 State C)
#if defined(ESP32)
#define EV_CHARGING LOW
#define EV_SUSPENDED HIGH
#elif defined(ESP8266)
#define EV_CHARGING HIGH
#define EV_SUSPENDED LOW
#endif

#define OCPP_AVAILABILITY_PIN 0 // Output pin | Signal if this EVSE is out of order (set by Central System)
#define OCPP_AVAILABLE HIGH
#define OCPP_UNAVAILABLE LOW

#define EVSE_GROUND_FAULT_PIN 13 // Input pin | Read ground fault detector
#define EVSE_GROUND_FAULTED HIGH
#define EVSE_GROUND_CLEAR LOW

/*
 * Pin mapping part II
 * Internal LED of ESP8266/ESP32 + additional ESP8266 development board LED
 */

#define SERVER_CONNECT_LED 16 // Output pin | Signal if connection to OCPP server has succeeded
#define SERVER_CONNECT_ON LOW
#define SERVER_CONNECT_OFF HIGH

#define CHARGE_PERMISSION_LED 2 // Output pin | Signal if OCPP allows / forbids energy flow
#if defined(ESP32)
[... 477 lines cut ...]
            uint16_t inputPort = 0;
            for (unsigned int j = i + 1; j < inputUrl.length(); j++) {
                if (isDigit(inputUrl.charAt(j))) {
                    inputPort *= (uint16_t) 10;
                    inputPort += inputUrl.charAt(j) - '0';
                } else {
                    break;
                }
            }
            if (inputPort > 0) {
                port = inputPort;
            }
            break;
        } else if (inputUrl.charAt(i) == '/') { //case host.com/path
            host = inputUrl.substring(0, i);
            break;
        }
    }
    if (host.isEmpty()) {
        host = inputUrl;
    }

    return true;
}
`

Fareed99, 2023-03-31:
@matth-x @piyush9999 I followed the above discussion to implement multiple connectors. Is there a way to call startTransaction and stopTransaction functions for a specific connector?
I'm aware the connectorId can be chosen from RemoteStartTransaction and RemoteStopTransaction but is there a way to do it with startTransaction and stopTransaction?

piyush9999, 2023-03-31:
Hello @Fareed99 you can try by giving a separate IDTAG, this should work

Fareed99, 2023-03-31:
Hey @piyush9999 thanks for your reply.
startTransaction and stopTransaction do not have the connectorId paramter. So whenever I call startTransaction, it starts a transaction on connector 1 but I also need to be able to start and stop transactions on connectors 2 and onwards.
@piyush9999 @matth-x Please shed some light on this, I would be grateful. Thanks!

Navjot0005, 2023-09-29:
@matth-x Hi hope you are good.
I am working on this lib and everything is ok till now. However, I want to send the RFID to Steve pls can you or anyone  tell me how can I do this??

piyush9999, 2023-09-29:
`if (/* RFID chip detected? */ false) {
        String idTag = "0123456789ABCD"; //e.g. idTag = RFID.readIdTag();

        if (!getTransaction()) {
            //no transaction running or preparing. Begin a new transaction
            Serial.printf("[main] Begin Transaction with idTag %s\n", idTag.c_str());

            /*
             * Begin Transaction. The OCPP lib will prepare transaction by checking the Authorization
             * and listen to the ConnectorPlugged Input. When the Authorization succeeds and an EV
             * is plugged, the OCPP lib will send the StartTransaction
             */
            auto ret = beginTransaction(idTag.c_str());

            if (ret) {
                Serial.println(F("[main] Transaction initiated. OCPP lib will send a StartTransaction when" \
                                 "ConnectorPlugged Input becomes true and if the Authorization succeeds"));
            } else {
                Serial.println(F("[main] No transaction initiated"));
            }
`
After reading the tag i.e idTag it will compare the tag you added on the server,

Navjot0005, 2023-09-29:
Thanks for the help @piyush9999 ..
Right now I am reading the tag and then copying this to the 32-bit variable.
but the thing is I could not send the read id tag to Steve.
Will it send the   "idTag = "0123456789ABCD";  to the steve server ??
Or how Can I send the RFID tag to steve??
Thanks for reply.

piyush9999, 2023-09-29:
Don't send 0123456789ABCD, add the rfid to the server first, for more comfort make it as parent idtag, make String idtag = RFID.readIdTag();

Navjot0005, 2023-10-03:
Hi @piyush9999.
Thanks again for guiding.
Actually Can you tell me how I can send the value because in my case it is required to send the read rfid tag to steve.
Also one thing more I want to add in this.
Like if load not connected then it send reject msg to steve and relays gets off.